coolsnowwolf/lede #5600: V2升级到1.30编译过不去了帮忙看看怎么整

stock169, 2020-10-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：新版v2 和 v2-plugin 都编译过不去，版本降回4.28.2 和1.4.3就能编译通过

### (2) 路由器型号和固件版本

A：P&W R619AC

### (3) 详细日志

A：v2ray.com/core/common/protocol/tls/cert
github.com/xtls/go
v2ray.com/core/transport/internet/tls
# github.com/xtls/go
pkg/mod/github.com/xtls/go@v0.0.0-20200921133830-416584838a0f/auth.go:29:7: undefined: ecdsa.VerifyASN1
pkg/mod/github.com/xtls/go@v0.0.0-20200921133830-416584838a0f/key_schedule.go:177:16: xShared.FillBytes undefined (type *big.Int has no field or method FillBytes)
v2ray.com/core/app/reverse
v2ray.com/core/app/proxyman/outbound
v2ray.com/core/app/stats
google.golang.org/grpc
v2ray.com/core/common/platform/ctlcmd
v2ray.com/core/common/platform/filesystem
v2ray.com/core/proxy/blackhole
v2ray.com/core/main/confloader/external
v2ray.com/core/proxy/dns
v2ray.com/core/proxy/dokodemo
v2ray.com/core/app/router
v2ray.com/core/common/retry
v2ray.com/core/proxy/mtproto
v2ray.com/core/proxy/freedom
v2ray.com/core/app/dns
v2ray.com/core/proxy/http
v2ray.com/core/proxy/shadowsocks
v2ray.com/core/app/commander
v2ray.com/core/app/log/command
v2ray.com/core/app/proxyman/command
v2ray.com/core/app/stats/command
v2ray.com/core/proxy/socks
v2ray.com/core/proxy/trojan
v2ray.com/core/proxy/vless
hash/crc64
v2ray.com/core/proxy/vless/encoding
github.com/dgryski/go-metro
github.com/seiflotfy/cuckoofilter
golang.org/x/crypto/sha3
v2ray.com/core/transport/internet/headers/http
v2ray.com/core/common/antireplay
v2ray.com/core/proxy/vmess/aead
v2ray.com/core/transport/internet/headers/noop
v2ray.com/core/transport/internet/headers/srtp
v2ray.com/core/proxy/vmess
v2ray.com/core/transport/internet/headers/tls
v2ray.com/core/transport/internet/headers/utp
v2ray.com/core/transport/internet/headers/wechat
v2ray.com/core/transport/internet/headers/wireguard
golang.org/x/net/http2/h2c
v2ray.com/core/proxy/vmess/encoding
v2ray.com/core/transport/internet/kcp
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/protocol
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/utils
v2ray.com/core/transport/internet/http
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/congestion
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/qerr
v2ray.com/core/proxy/vmess/inbound
v2ray.com/core/proxy/vmess/outbound
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/wire
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/flowcontrol
v2ray.com/core/external/github.com/cloudflare/sidh/sidh
github.com/gorilla/websocket
v2ray.com/core/infra/conf/json
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/ackhandler
v2ray.com/core/external/github.com/marten-seemann/qtls
v2ray.com/core/transport/internet/websocket
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/handshake
v2ray.com/core/external/github.com/lucas-clemente/quic-go
v2ray.com/core/transport/internet/quic

Cleaning module download cache (golang/go#27455)

Makefile:308: recipe for target '/home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/v2ray-core-4.30.0/.built' failed
make[3]: *** [/home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/v2ray-core-4.30.0/.built] Error 2
make[3]: Leaving directory '/home/pchome/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#96.80#12.79#45.82
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/pchome/lede'
package/Makefile:107: recipe for target '/home/pchome/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/pchome/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pchome/lede'
/home/pchome/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Ljzkirito, 2020-10-04:
有没有./scripts/feeds update -a && ./scripts/feeds install -a
新版v2 requires Go 1.15+，你可能是feeds未更新

stock169, 2020-10-04:
> 有没有./scripts/feeds update -a && ./scripts/feeds install -a
> 新版v2 requires Go 1.15+，你可能是feeds未更新

还真是这个问题，是我疏忽了feeds更新是做过的但我没看提示实际更新失败的，现在搞定了非常感谢。

closed 2021-06-20 by coolsnowwolf

oldcat618, 2021-11-03:
GO_VERSION_MAJOR_MINOR:=1.16
GO_VERSION_PATCH:=9

PKG_HASH:=0a1cc7fd7bd20448f71ebed64d846138850d5099b18cf5cc10a4fc45160d8c3d
更换这个go版本通过