coolsnowwolf/lede #5601: 编译失败，怎么修改都不行，崩溃中。。。

recinfo, 2020-10-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

编译失败，崩溃了，再解决不了就不装naiveproxy这个了；如果不装编译就没问题。

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

make[3]: Entering directory '/home/hp/lede/package/openwrt-packages/naiveproxy'
rm -f /home/hp/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.built
touch /home/hp/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3/.built_check
( cd /home/hp/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-3 ; ./tools/import-upstream.sh ; . ./init_env.sh "x86_64" "x86" "" "" "/home/hp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl" "/home/hp/lede/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/home/hp/lede/staging_dir/target-x86_64_musl" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/home/hp/lede/staging_dir/hostpkg/bin/python3" ; /home/hp/lede/staging_dir/hostpkg/bin/ninja -C "out/Release" naive ; )
+ cut -d= -f2 src/chrome/VERSION
+ tr \n .
+ cut -d. -f1-4
+ have_version=85.0.4183.83
+ cat CHROMIUM_VERSION
+ want_version=85.0.4183.83
+ [ 85.0.4183.83 = 85.0.4183.83 ]
+ exit 0
Done. Made 12712 targets from 2280 files in 59184ms
ninja: Entering directory `out/Release'
[1/1922] CXX obj/buildtools/third_party/libc++/libc++/memory.o
FAILED: obj/buildtools/third_party/libc++/libc++/memory.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/memory.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=__attribute__\(\(__visibility__\(\"default\"\)\)\) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/blacklist.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -fstrict-aliasing -fPIC -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/hp/lede/staging_dir/target-x86_64_musl/usr/include -I/home/hp/lede/staging_dir/target-x86_64_musl/usr/include/nss -I/home/hp/lede/staging_dir/target-x86_64_musl/usr/include/nspr -fvisibility-inlines-hidden --target=x86_64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../buildtools/third_party/libc++/trunk/src/memory.cpp -o obj/buildtools/third_party/libc++/libc++/memory.o
Bus error (core dumped)
[2/1922] CXX obj/base/base_static/base_switches.o
FAILED: obj/base/base_static/base_switches.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/base/base_static/base_switches.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/blacklist.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fno-ident -fdata-sections -ffunction-sections -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/hp/lede/staging_dir/target-x86_64_musl/usr/include -I/home/hp/lede/staging_dir/target-x86_64_musl/usr/include/nss -I/home/hp/lede/staging_dir/target-x86_64_musl/usr/include/nspr -fvisibility-inlines-hidden --target=x86_64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../base/base_switches.cc -o obj/base/base_static/base_switches.o
Bus error (core dumped)
[3/1922] CXX obj/buildtools/third_party/libc++/libc++/mutex.o
FAILED: obj/buildtools/third_party/libc++/libc++/mutex.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/mutex.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=__attribute__\(\(__visibility__\(\"default\"\)\)\) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n356902-4e813bbd-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=d6879d611f3dcf3da8dd92e080029394aa30bc42 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 [... cut ...]
package/Makefile:111: recipe for target 'package/openwrt-packages/naiveproxy/compile' failed
make[2]: *** [package/openwrt-packages/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/hp/lede'
package/Makefile:107: recipe for target '/home/hp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/hp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hp/lede'
/home/hp/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Dimitri2020007, 2020-10-04:
跳过naiveproxy的编译

Myoko, 2020-10-04:
略过naiveproxy 不要编译这个

ligf, 2020-10-17:
我的解决办法
1.用https://github.com/xiaorouji/openwrt-package/tree/master/package这里面的naiveproxy替换helloworld的包
2.安装sudo apt-get install libtinfo5
3.增加编译机器内存，我改成4G后可以正常编译

UJStudio, 2020-10-22:
经过验证
sudo apt-get install libtinfo5
解决问题。

lxwq, 2020-10-23:
sudo apt-get install libtinfo5
安装后编译通过,谢谢,

closed 2021-06-20 by coolsnowwolf