coolsnowwolf/lede #5602: 使用img写盘工具出现问题

recinfo, 2020-10-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

使用img写盘工具出现问题。成功编译了好几个均出现该问题。

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
使用 IMG写盘工具 往u盘写入编译成功的img文件，在最后提示该错误，但不影响使用。我是在ubuntu18.04物理机系统下编译并另存到win系统NTFS格式盘中，在win系统下写盘。
### (1) 

A：

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：**Error #0 occured while writting to disk at sector 361472**

hcym, 2020-10-04:
ubu实体机直接dd不是更方便吗，win下面写盘都是刷手机刷安卓，

9l, 2020-10-04:
可能，你的u盘寿命不行了

qianxu2001, 2020-10-04:
换个软件不就行了，Roadkil's DiskImage 都多老了，用 Etcher 就没这么多问题

Pastechn, 2020-10-04:
同出现这个问题 可以确定硬盘和U盘没有任何问题 5.4内核的会出现 4.19没有问题

sekaiacg, 2020-10-04:
同样的问题，其他固件正常

Dimitri2020007, 2020-10-05:
使用etcher或者rufus写盘，推荐rufus

Ljzkirito, 2020-10-05:
Roadkil's DiskImage 写5.4内核的会出现 4.19没有问题，还是换个工具吧。
Dimitri2020007 说的[Rufus](https://rufus.ie/)很好用，支持中文，uefi兼容性好

adminpomme, 2020-10-06:
PE 下面用physdiskwrite写盘,没有出过问题

hcym, 2020-10-06:
rufus不错的，我都是写win10和Ubuntu，op还没用过

Dimitri2020007, 2020-10-06:
rufus写盘把高级选项的内置硬盘打开就可以写了，rufus支持系统应该也支持op。选dd模式写入就行。

closed 2020-10-12 by recinfo