coolsnowwolf/lede #5604: 竞斗云2.0无法开启9984的无线

mankitlau2017, 2020-10-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

竞斗云2.0无法开启9984的无线

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

最新源码编译p2w-r619ac-128m，修改了qcom-ipq4019-r619ac.dtsi文件，添加了pcie的定义，编译时勾选了9984-ct的驱动，刷入后开机成功识别网卡，手动补齐art后，成功读取正确的mac，但是无法启动无线。

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

型号：P&W R619AC 128M
固件版本：OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)
内核版本：4.19.138

### (3) 详细日志

附上启动日志：
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.138 (mkcp@mkcompile) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2882-4abf84eae)) #0 SMP Wed Sep 16 12:04:43 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: P&W R619AC 128M
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x438 with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu s30156 r8192 d23092 u61440
[    0.000000] pcpu-alloc: s30156 r8192 d23092 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 rootfstype=squashfs
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 508276K/522240K available (4920K kernel code, 181K rwdata, 1336K rodata, 1024K init, 235K bss, 13964K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (5913 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 182 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 236 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000009] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000023] Switching to timer-based delay loop, resolution 20ns
[    0.000252] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000269] pid_max: default: 32768 minimum: 301
[    0.000456] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000474] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001315] CPU: Testing write buffer coherency: ok
[    0.002281] Setting up static identity map for 0x80300000 - 0x80300060
[... 424 lines cut ...]
[1446972.240795] ath10k_pci 0000:01:00.0: firmware crashed! (guid n/a)
[1446972.247524] ath10k_pci 0000:01:00.0: Could not init hif: -5
[1446972.259583] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[1446972.265470] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[1446972.280793] ath10k_pci 0000:01:00.0: firmware ver 10.4-3.9.0.2-00070 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps,peer-fixed-rate crc32 873782fb
[1446972.285032] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 85498734
[1446972.299674] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[1446972.307301] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 4: -28
[1446972.317532] ath10k_pci 0000:01:00.0: failed to read firmware dump area: -28
[1446972.326437] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[1446972.333638] ath10k_pci 0000:01:00.0: [00]: 0x0004a000 4294967295 4294967295 4294967295 4294967295
[1446972.339756] ath10k_pci 0000:01:00.0: [01]: 0x0004a400 4294967295 4294967295 4294967295 4294967295
[1446972.348595] ath10k_pci 0000:01:00.0: [02]: 0x0004a800 4294967295 4294967295 4294967295 4294967295
[1446972.357668] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00 4294967295 4294967295 4294967295 4294967295
[1446972.366673] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 4294967295 4294967295 4294967295 4294967295
[1446972.375639] ath10k_pci 0000:01:00.0: [05]: 0x0004b400 4294967295 4294967295 4294967295 4294967295
[1446972.384743] ath10k_pci 0000:01:00.0: [06]: 0x0004b800 4294967295 4294967295 4294967295 4294967295
[1446972.393740] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00 4294967295 4294967295 4294967295 4294967295
[1446972.402765] ath10k_pci 0000:01:00.0: [08]: 0x0004c000 4294967295 4294967295 4294967295 4294967295
[1446972.411827] ath10k_pci 0000:01:00.0: [09]: 0x0004c400 4294967295 4294967295 4294967295 4294967295
[1446972.420822] ath10k_pci 0000:01:00.0: [10]: 0x0004c800 4294967295 4294967295 4294967295 4294967295
[1446972.429882] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00 4294967295 4294967295 4294967295 4294967295
[1446972.438912] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 8: -28
[1446972.447896] ath10k_pci 0000:01:00.0: failed to dump debug log area: -28
[1446972.507900] ath10k_pci 0000:01:00.0: cannot restart a device that hasn't been started

AmadeusGhost, 2020-11-06:
https://github.com/AmadeusGhost/lede/commit/7159a192582b876eb667219819a2136d3b8cbece 实测 5.4 内核下正常使用

Gh-mean, 2021-03-26:
我也遇到了，但是我是可以开wifi的。只是系统报错：kern.info kernel: [  244.134856] ath10k_pci 0000:01:00.0: mac flush null vif, drop 0 queues 0xffff

closed 2021-06-20 by coolsnowwolf