coolsnowwolf/lede #5606: 编译不成功大概是什么问题

ratywang, 2020-10-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译出错

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：ModuleNotFoundError: No module named '_ctypes'
Makefile:101: recipe for target '/home/raty/lede/build_dir/target-x86_64_musl/zstd-1.4.5/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/raty/lede/build_dir/target-x86_64_musl/zstd-1.4.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/raty/lede/feeds/packages/utils/zstd'
time: package/feeds/packages/zstd/compile#0.89#0.16#5.43
package/Makefile:111: recipe for target 'package/feeds/packages/zstd/compile' failed
make[2]: *** [package/feeds/packages/zstd/compile] Error 2
make[2]: Leaving directory '/home/raty/lede'
package/Makefile:107: recipe for target '/home/raty/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/raty/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/raty/lede'
/home/raty/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Dimitri2020007, 2020-10-06:
软件包在编译前下载完整了吗？

ratywang, 2020-10-06:
好像没有报错，helloworld的源加上就报错，不加就没问题。奇怪了。

Dimitri2020007, 2020-10-06:
是不是编译了naiveproxy，那个是有bug的。
你的报错信息提示是ctypes这个模块的名字没找到

make clean后再编译一次，是脚本就重新开。

Myoko, 2020-10-06:
不要尝试去编译 naiveproxy

Sheep-12, 2020-10-06:
naiveproxy 暂且放一放

Dimitri2020007, 2020-10-07:
编译了有冲突的kmod软件包、如固件firmware、无线网卡驱动kmod、网络支持network support、实用程序utlities中的encryption、compression与busybox中重复、tar、network中的ss/ssr-libev、network下ipaddress开头下的unbound、network下的iperf3（选了iperf3就不要选iperf3-ssl。）多数是功能重复引发的冲突。

调整分区空间太大，编译机硬盘空间不足出现的失败。

还有少数是源码软件有bug或已经停更放弃的。有luci-app-cjdns（最后安装ipk时出现argument list too long、还有一个kmod-batman也是一样的。）、luci-app-acme、luci-app-cshark、luci-app-olsr*、naiveproxy、v2ray？（不知道有没有即时更新最新v2ray-core和sum）。

若选了kcptun、obfs4proxy、v2raygo、unblockneteasemusic的go模块则会在编译中使用go网络下载go模块，如果线路不好是很容易失败的。解决方法在用户的~/.bashrc中配置goproxy变量。中国大陆配置goproxy.cn或者goproxy.io（哪个又快又稳定用哪个）

实际编译不成功的原因碰到的目前只有这些了。

up或油管主是不会提到如此详细的。

实体+脚本编译1个半月的亲自体验。

closed 2021-06-20 by coolsnowwolf