coolsnowwolf/lede #5609: 编译成功，也出固件了，但是刷入无限重启

royalwei, 2020-10-06:
x64软路由，开机跑码一段时间后出现reboot in 1s，直接重启了，是不是选了什么不该选的软件包？

AmadeusGhost, 2020-10-06:
你倒是贴开机跑码的 log 啊，不难鬼知道啥问题

royalwei, 2020-10-06:
> 跑码的 log 啊，不难鬼知道啥
老铁我也想，但是太快了

royalwei, 2020-10-06:
[image]

royalwei, 2020-10-06:
[image]

AmadeusGhost, 2020-10-06:
啥硬盘啥文件系统？

royalwei, 2020-10-06:
就是写盘工具直接写的

AmadeusGhost, 2020-10-06:
vm 的话不应该是用 vmdk 嘛

royalwei, 2020-10-06:
> vm 的话不应该是用 vmdk 嘛

是的，pc太难截到，所以我用vm复现，vm用的编译目录下生成的vmdk，pc用的combine固件

Dimitri2020007, 2020-10-06:
俺觉得你固件分区太小了才会无限重启的。
若不是固件分区太小无限重启
使用linux下的qemu-img命令转盘

qemu-img -f raw #原盘名字 -O vmdk #转换固件的名字.vmdk

转出来直接添加进虚拟机就行了啊，无需写盘。

Dimitri2020007, 2020-10-07:
转盘命令是qemu-img convert -f raw -O vmdk #原固件名 #目标固件名.vmdk

Promix953, 2020-10-08:
config文件发下看看，最好能截取List of all partitions:之前的log

a13547, 2020-10-11:
重新编译一下，把kernel和root的容量都调大点，选择ext4镜像刷入应该就可以了。有段时间官方的squash固件也出现了同样的问题。

royalwei, 2020-10-25:
> 重新编译一下，把kernel和root的容量都调大点，选择ext4镜像刷入应该就可以了。有段时间官方的squash固件也出现了同样的问题。

大佬真心求教，网上搜不到教程

royalwei, 2020-10-25:
> 重新编译一下，把kernel和root的容量都调大点，选择ext4镜像刷入应该就可以了。有段时间官方的squash固件也出现了同样的问题。

谢谢，找到了！！！

closed 2020-10-25 by royalwei