coolsnowwolf/lede #5613: 麻烦哪位大神, 帮我解答一些关于，openwrt 应用 python出现错误的问题，非常感谢🙏

jump-to-jump, 2020-10-06:
**具体问题**

A：我的**树莓派4**现在安装了一个散热风扇，它需要通过l2c的接口来控制风扇的运作，我已经安装有python的驱动文件和i2c的驱动，当我用openwrt的驱动运行.py的python文件时 他出现了错误无法运行。

**错误代码**
Traceback (most recent call last):
       File "/mnt/mmcblk0p1/fan/fan.py", line 3, in <module>
             bus = sambus.SMBus(1)
FileNotFoundError: [Errno 2] No such file or directory

**我现在openwrt安装了l2c 的所有固件**
Kernel modules -> I2C Support -> kmod-i2c-algo—pca          # 树莓派 i2c  PCA 9564 interfaces
Kernel modules -> I2C Support -> kmod-i2c-algo—pcf          # 树莓派 i2c  PCF 8584 interfaces
Kernel modules -> I2C Support -> kmod-i2c-bcm2835           # 树莓派 i2c  Broadcom BCM2835 I2C master controller driver
Kernel modules -> I2C Support -> kmod-i2c-core              # 树莓派 i2c 指针40指针硬件支持
Kernel modules -> I2C Support -> kmod-i2c-gpio              # 树莓派 i2c  GPIO-based bitbanging I2C
Kernel modules -> I2C Support -> kmod-i2c-mux               # 树莓派 i2c 指针bus固件支持 bus multiplexing support  
Kernel modules -> I2C Support -> kmod-i2c-mux-pca9541       # 树莓派 i2c 指针bus固件支持 Philips PCA9541 I2C mux/switches
Kernel modules -> I2C Support -> kmod-i2c-mux-pca954x       # 树莓派 i2c 指针bus固件支持 Philips PCA954x I2C mux/switches 
Kernel modules -> I2C Support -> kmod-i2c-pxa               # 树莓派 i2c 指针bus固件支持 Intel PXA I2C bus driver
Kernel modules -> I2C Support -> kmod-i2c-sambus            # 树莓派 i2c 指针bus固件支持 SMBus-specific protocols helper
Kernel modules -> I2C Support -> kmod-i2c-tiny-usb          # 树莓派 i2c 指针bus固件支持 I2C Tiny USB adaptor   

**Python的所有已安装的固件**
Languages -> Python -> docker-compose                     # Docker Compose
Languages -> Python -> python-pip-conf                    # configure file for pip/pip3
Languages -> Python -> python3                            # python 3 语言支持
Languages -> Python -> python-asyncio-src                 # Python 3.8 asyncio module (sources)
Languages -> Python -> python3-sambus                     # sambus库支持

**其他部分插件**
Utilities -> i2c-tools                                   # I2C tools for Linux
Utilities -> kmod                                       # Linux kernel module handling (tools)

麻烦有懂的大神解决一下，非常的感谢🙏

sxx1314, 2020-10-07:
如果用python等调试控制外部io。
还是用armbian之类的宿主系统，再docker一个openwrt即可。  镜像可以自己制作。
openwrt上折腾io非常不舒服，当然你也可以反方向在驱动都编译齐全的情况下把外部io共享给docker。这样你可以保留openwrt为宿主。

jump-to-jump, 2020-10-07:
> 如果用python等调试控制外部io。
> 还是用armbian之类的宿主系统，再docker一个openwrt即可。 镜像可以自己制作。
> openwrt上折腾io非常不舒服，当然你也可以反方向在驱动都编译齐全的情况下把外部io共享给docker。这样你可以保留openwrt为宿主。

jump-to-jump, 2020-10-08:
感谢sxx1314, 我试试

closed 2021-06-20 by coolsnowwolf