coolsnowwolf/lede #5614: r2s    WAN获取不到ip，源码也是重新搞得还是不行

dl250, 2020-10-07:
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Thu Jan 21 16:50:15 2016 kern.notice kernel: [    0.000000] Linux version 5.4.68 (ubuntu@OpenWrt) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2907-71e335c0b)) #0 SMP PREEMPT Fri Oct 2 16:41:31 2020
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] Machine model: FriendlyElec NanoPi R2S
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
Thu Jan 21 16:50:15 2016 kern.debug kernel: [    0.000000] On node 0 totalpages: 261632
Thu Jan 21 16:50:15 2016 kern.debug kernel: [    0.000000]   DMA32 zone: 4088 pages used for memmap
Thu Jan 21 16:50:15 2016 kern.debug kernel: [    0.000000]   DMA32 zone: 0 pages reserved
Thu Jan 21 16:50:15 2016 kern.debug kernel: [    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] percpu: Embedded 21 pages/cpu s45784 r8192 d32040 u86016
Thu Jan 21 16:50:15 2016 kern.debug kernel: [    0.000000] pcpu-alloc: s45784 r8192 d32040 u86016 alloc=21*4096
Thu Jan 21 16:50:15 2016 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
Thu Jan 21 16:50:15 2016 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] Memory: 1005264K/1046528K available (7294K kernel code, 520K rwdata, 2280K rodata, 1600K init, 702K bss, 33072K reserved, 8192K cma-reserved)
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] rcu: 	RCU event tracing is enabled.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] 	Tasks RCU enabled.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] GIC: Using split EOI/Deactivate mode
Thu Jan 21 16:50:15 2016 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x398 with crng_init=0
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.001335] Console: colour dummy device 80x25
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.001858] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.002790] pid_max: default: 32768 minimum: 301
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.003441] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.004122] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.006967] ASID allocator initialised with 32768 entries
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.007583] rcu: Hierarchical SRCU implementation.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.009146] smp: Bringing up secondary CPUs ...
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.010334] Detected VIPT I-cache on CPU1
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.010420] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.011170] Detected VIPT I-cache on CPU2
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.011227] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.011934] Detected VIPT I-cache on CPU3
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.011986] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.012098] smp: Brought up 1 node, 4 CPUs
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.015401] SMP: Total of 4 processors activated.
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.015834] CPU features: detected: 32-bit EL0 Support
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.016303] CPU features: detected: CRC32 instructions
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.024995] CPU: All CPU(s) started at EL2
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.025399] alternatives: patching kernel code
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.037445] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.038403] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.039661] pinctrl core: initialized pinctrl subsystem
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.041181] NET: Registered protocol family 16
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.043896] DMA: preallocated 256 KiB pool for atomic allocations
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.045522] cpuidle: using governor menu
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.046235] Serial: AMBA PL011 UART driver
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.072616] HugeTLB registered 1.00 GiB page size, pre-allocated 0 pages
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.073238] HugeTLB registered 32.0 MiB page size, pre-allocated 0 pages
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.073844] HugeTLB registered 2.00 MiB page size, pre-allocated 0 pages
Thu Jan 21 16:50:15 2016 kern.info kernel: [    0.074451] HugeTLB registered 64.0 KiB page size, pre-allocated 0 pages
[... 357 lines cut ...]
Thu Jan 21 16:51:49 2016 daemon.notice netifd: Interface 'wan' is disabled
Thu Jan 21 16:51:50 2016 kern.info kernel: [  103.755986] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [Generic PHY]
Thu Jan 21 16:51:50 2016 daemon.notice netifd: Interface 'wan' is enabled
Thu Jan 21 16:51:50 2016 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jan 21 16:51:50 2016 kern.info kernel: [  103.767810] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
Thu Jan 21 16:51:50 2016 kern.warn kernel: [  103.768505] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[... cut ...]
Thu Jan 21 16:51:55 2016 kern.info kernel: [  109.020820] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Jan 21 16:51:55 2016 kern.info kernel: [  109.021672] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Thu Jan 21 16:51:55 2016 daemon.notice netifd: wan (1883): udhcpc: started, v1.31.1
Thu Jan 21 16:51:55 2016 daemon.notice netifd: wan (1883): udhcpc: sending discover
Thu Jan 21 16:51:58 2016 daemon.notice netifd: wan (1883): udhcpc: sending discover
Thu Jan 21 16:52:01 2016 daemon.notice netifd: wan (1883): udhcpc: sending discover

1715173329, 2020-10-07:
[image]
this is a duplicate of #5590.

dl250, 2020-10-07:
我试过了不行，我现在把源码回退到以前去

dl250, 2020-10-07:
源码我回退到https://github.com/coolsnowwolf/lede/commit/9a482a15a3d14849b050ab7556ab9542b47b2c21 这里也就是24号的，编译出来正常没问题

hxlls, 2020-10-07:
有修复的PR 提交，据说已经修复，但是没合并，也就没测试

467815891a, 2020-10-08:
今天已经修复了，git pull之后重新编译即可。

joesdu, 2020-10-08:
就很奇怪,我拉了最新的代码编译后,直接开不了机.

joesdu, 2020-10-08:
真的是不能追新,新的代码固件编译后直接无法启动.用我之前8月2号编译的版本就是好的,没名字显示就没名字显示吧.😂

hxlls, 2020-10-08:
推送的测试可以了，但是。。。我发现不懂是天生R2S和这个不合还是怎么的，运行一段时间后莫名其妙的的挂了，PING 不通那种。。也无法上网。。醉了，用友善的那个WRT 合体的就没事，好久了，一直没找到原因。。蛋疼

zgc, 2020-10-09:
> 就很奇怪,我拉了最新的代码编译后,直接开不了机.

同样的问题

asushugo, 2020-10-09:
> > 就很奇怪,我拉了最新的代码编译后,直接开不了机.
> 
> 同样的问题

没错，我昨晚测试了一晚上，我还以为是我自己合并出错了，但是重新拉代码编译还是无法启动

1715173329, 2020-10-09:
https://t.me/nanopi_r2s
试一下这个频道里面发布的固件

rongcolor, 2020-10-09:
我clone重新拉的L大的，没有pull，编译出来的没有问题。正常

rongcolor, 2020-10-09:
[image]

zgc, 2020-10-10:
> [t.me/nanopi_r2s](https://t.me/nanopi_r2s)
> 试一下这个频道里面发布的固件

尝试了openwrt-18.06-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz这个固件，能启动，进不去后台,分配的ip是169.254.124.63，指定ip也进不去

whalechoi, 2020-10-10:
刚刚拉的最新代码，不保存配置刷的，直接无法开机了，灯亮了一会儿，然后熄了

closed 2021-06-20 by coolsnowwolf