coolsnowwolf/lede #5615: 编译最新版拨号后无法上网

759071793, 2020-10-07:
最新版5.4.68内核编译后无法打开网页，貌似是客户机没有获取到dns，客户机自定义dns也打不开网页，必须打开Turbo ACC 网络加速里的dns加速才能打开网页，不知道为什么，然后打开dns加速之后，开启frp，重启软路由后frp就无法连接到服务端了，现实time out，回到我之前编译的5.4.61内核的就什么都正常，好奇怪，有大佬解答下吗，这个问题不知道贴哪里的日志。

759071793, 2020-10-07:
我发现最新版的内核日志会跑出一个https-dns-proxy的列表，以前的内核是没有这个日志的，是最新加入的吗？貌似是这个问题，请问怎么取消呢
[image]

759071793, 2020-10-07:
可能是编译lienol的passwall插件的问题吧，但是奇怪的是我刚编译好并没有启用passwall啊，算了此贴终结

closed 2020-10-07 by 759071793

Myoko, 2020-10-07:
暂时还是用4.19吧！

Myoko, 2020-10-07:
暂时还是用4.19吧！

759071793, 2020-10-07:
> 暂时还是用4.19吧！

没使用过这个内核，我也是刚开始学习编译这个，一开始使用的是5.4.61

Myoko, 2020-10-07:
> > 暂时还是用4.19吧！
> 
> 没使用过这个内核，我也是刚开始学习编译这个，一开始使用的是5.4.61

https://github.com/coolsnowwolf/openwrt

759071793, 2020-10-07:
> > > 暂时还是用4.19吧！
> > 
> > 
> > 没使用过这个内核，我也是刚开始学习编译这个，一开始使用的是5.4.61
> 
> 大雕的稳定版仓库
> https://github.com/coolsnowwolf/openwrt

感谢！我刚刚去掉lienol的passwall插件目前完全正常了