coolsnowwolf/lede #5617: 编译错误

linxuankent, 2020-10-07:
make[6]: Entering directory '/home/linxuan/snap/ceshi/lede/build_dir/target-x86_64_musl/olsrd-2020-06-18-84400e5c/lib/pud/nmealib'
[CC] src/context.c
[CC] src/generator.c
[CC] src/gpgga.c
[CC] src/gpgsa.c
[CC] src/gpgsv.c
[CC] src/gprmc.c
[CC] src/gpvtg.c
[CC] src/info.c
[CC] src/nmath.c
[CC] src/parser.c
[CC] src/sentence.c
[CC] src/util.c
[CC] src/validate.c
[AR] lib/libnmea.a
[LD] lib/libnmea.so
make[6]: Leaving directory '/home/linxuan/snap/ceshi/lede/build_dir/target-x86_64_musl/olsrd-2020-06-18-84400e5c/lib/pud/nmealib'
make[6]: Entering directory '/home/linxuan/snap/ceshi/lede/build_dir/target-x86_64_musl/olsrd-2020-06-18-84400e5c/lib/pud/wireformat'
[CC] src/nodeIdConversion.c
[CC] src/wireFormat.c
[LD] lib/libOlsrdPudWireFormat.so
make[6]: Leaving directory '/home/linxuan/snap/ceshi/lede/build_dir/target-x86_64_musl/olsrd-2020-06-18-84400e5c/lib/pud/wireformat'
[CC] src/configTools.c
[CC] src/configuration.c
[CC] src/dedup.c
[CC] src/gpsConversion.c
[CC] src/gpsdclient.c
src/gpsdclient.c: In function 'nmeaInfoFromGpsd':
src/gpsdclient.c:373:20: error: 'struct gps_fix_t' has no member named 'status'; did you mean 'datum'?
   if (gpsdata->fix.status == STATUS_NO_FIX) {
                    ^~~~~~
                    datum
make[5]: *** [../../Makefile.inc:358: src/gpsdclient.o] Error 1
make[5]: Leaving directory '/home/linxuan/snap/ceshi/lede/build_dir/target-x86_64_musl/olsrd-2020-06-18-84400e5c/lib/pud'
make[4]: *** [Makefile:251: libs] Error 2
make[4]: Leaving directory '/home/linxuan/snap/ceshi/lede/build_dir/target-x86_64_musl/olsrd-2020-06-18-84400e5c'
make[3]: *** [Makefile:296: /home/linxuan/snap/ceshi/lede/build_dir/target-x86_64_musl/olsrd-2020-06-18-84400e5c/.built] Error 2
make[3]: Leaving directory '/home/linxuan/snap/ceshi/lede/feeds/routing/olsrd'
time: package/feeds/routing/olsrd/compile#10.88#4.28#1202.53
make[2]: *** [package/Makefile:113: package/feeds/routing/olsrd/compile] Error 2
make[2]: Leaving directory '/home/linxuan/snap/ceshi/lede'
make[1]: *** [package/Makefile:107: /home/linxuan/snap/ceshi/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/linxuan/snap/ceshi/lede'
make: *** [/home/linxuan/snap/ceshi/lede/include/toplevel.mk:224：world] 错误 2

尝试删除重新下载包后还是错误

Dimitri2020007, 2020-10-08:
是否使用gitee克隆lede源码加速了？

linxuankent, 2020-10-08:
> （使用gitee
> 复制lede源码减少git clone时间）
> 
> 编译模块时候会产生莫名奇妙的错误。
> olsrd就是我碰到的其中一个啊
> 
> 重新从github复制
> lede的源码不要使用gitee再试一次

正常git 源码，没使用gitee

hyird, 2020-10-09:
官方 为了兼容新版本的一个库， 打了个patch 删掉即可 

rm -rf package/feeds/routing/olsrd/patches/012-gpsd.patch

linxuankent, 2020-10-10:
> 官方为了兼容新版本的一个库，打了个补丁
> 
> rm -rf package / feeds / routing / olsrd / patches / 012-gpsd.patch

可以了

closed 2020-10-10 by linxuankent