coolsnowwolf/lede #5619: o 为什么我编译的固件 本地千兆用IPERF3只能跑到500不到..用其它网友编译的.可以跑到930

showme2005, 2020-10-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我自己编译的固件,本地速度跑不满千兆，只有500不到，但是用其它网友编译的 ，可以跑到930.

### (2) 路由器型号和固件版本

A：asus-acrh17, R20.9.15,

### (3) 详细日志

A：无日志,用iperf3 测试自己本地的吞吐量..500不到,换成其它网友的固件可以跑到930..

ggllioio, 2020-10-07:
我mt7621a最新源码编译也遇到类似问题。
楼主看下跑iperf是不是cpu单核占用100%，其它核心闲置？

Dimitri2020007, 2020-10-07:
编译了对应型号的网卡固件和komd驱动吗？
是不是无线测速的？

showme2005, 2020-10-08:
> 编译了对应型号的网卡固件和kmod驱动吗？
> 是不是无线测速的？
无线测速更惨 300不到...我是用的有线测试的  500不到...我还是用的5G信号..我这破路由开不了 160..就只能开到80...

showme2005, 2020-10-08:
> 我mt7621a最新源码编译也遇到类似问题。
> 楼主看下跑iperf是不是cpu单核占用100%，其它核心闲置？

这个和CPU有关系 ??路由器上开iperf3 -s..然后笔记本和电脑都用iperf3.exe -c  路由器IP地址　　测试...有线 500不到..无线 300不到..好惨..换一个固件,无线没有什么变化,但是有线就可以正常到930

AmadeusGhost, 2020-10-08:
路由器上开 iperf3 server 并不是测的 nat 性能啊

showme2005, 2020-10-09:
> 路由器上开 iperf3 server 并不是测的 nat 性能啊
这个肯定不是那种特别专业的。。。只是反映一个情况。。。内网之间传文件速度达不到100M。

hyird, 2020-10-09:
5.4内核上的无线驱动很拉垮 5g 单核吃满才跑400+  
用li大的4.14内核的可以跑600+。。。
还是刷集客吧 1.2g没问题

closed 2021-06-20 by coolsnowwolf