coolsnowwolf/lede #5620: 小娱c1（mt7621a）wan to lan开sfe只有400多M

ggllioio, 2020-10-07:
Q：是否搜索了issue (使用 "x" 选择)

有arch17反应这个问题。

## 2. 详细叙述

小娱路由使用昨天最新源码编译，只编译了基本路由功能（带ipv6）。
关闭sfe等加速，测试跑300m；开sfe，能跑不到500m。
期间用htop监测，cpu单线程占用100%。其它3线程基本空闲。
测试网友的x-wrt和padavan，软nat也能800-900M。

### (2) 路由器型号和固件版本

A：小娱c1/c5

### (3) 详细日志

A：无日志

AmadeusGhost, 2020-10-08:
试试用  flow offload 和 hwnat

ggllioio, 2020-10-08:
> 请用flow offload和hwnat

使用flow offload和hwnat也测试了。单开flow也不到500m，单核占满100%。开hwnat后重启测试，速度不变，似乎并没有生效。

AmadeusGhost, 2020-10-08:
> 使用flow offload和hwnat也测试了。单开flow也不到500m，单核占满100%。开hwnat后重启测试，速度不变，似乎并没有生效。

最新代码？

ggllioio, 2020-10-08:
> > 使用flow offload和hwnat也测试了。单开flow也不到500m，单核占满100%。开hwnat后重启测试，速度不变，似乎并没有生效。
> 
> 最新代码？

最新代码。并且刚才试了下用官方openwrt编译也这样。单核100%跑永远不过500。hwnat不生效。

AmadeusGhost, 2020-10-08:
> 最新代码。并且刚才试了下用官方openwrt编译也这样。单核100%跑永远不过500。hwnat不生效。

emmm，不应该啊，官方现在用的是 DSA 驱动，跑不上去正常，小娱 c5 的已经改回旧驱动了，正常来说 hwnat 应该能工作
有空再看看啥情况吧，暑假那时候试过 hwnat 是能正常工作的

ggllioio, 2020-10-08:
.........问题似乎解决了......我一般习惯把luci-app-statistics这个便以上，使用collectd监测cpu，内存，负载，流量等。
把这个去掉网速似乎恢复正常了。能跑接近1000.
但是日常使用还是只有一个核心工作。

AmadeusGhost, 2020-10-08:
所以至少 hwnat 能正常工作嘛？😂

closed 2021-06-20 by coolsnowwolf