coolsnowwolf/lede #5625: adguardhome核心版本不显示版本号，怎么都是空白

recinfo, 2020-10-08:
全新安装后更新核心，更新完之后无法获取核心版本号，使用的是LuCI Master (git-20.256.12360-1a54222) / OpenWrt R20.9.15。ADG版本为1.8-20200824

更新后正常下载，然后提示下面的东西：
Local version: v0.103.3, cloud version: v0.103.3.
Succeeded in updating core.
luci enable switch=1
AdGuardHome service enabled
workdir is a ext4 filesystem
**Command failed: Not found** 这里不知道有没有影响？？
AdGuardHome service disabled
start copy
download success start copy

我看应该是成功了，管理后台也能进入，AdGuardHome.yaml文件也有，可是核心版本的位置就是不显示版本号，也没有报错，显示空白。在ADG功能启动和停止的状态下都试过，均是如此，反复重启也没用，是我哪里搞错了么？

FusionPlmH, 2020-10-08:
問錯地方

recinfo, 2020-10-08:
感谢回复，请关闭。

顺便说一下，我突发奇想感觉会不会是theme的问题，经过测试果然是，更换其他的theme就能正常显示。仅在win7下测试是这样，其他系统不保证。

Aliceliese, 2020-10-08:
建議自己手動關閉，不用等其他人來

mysteryman2045, 2020-10-09:
> 感谢回复，请关闭。
> 
> 顺便说一下，我突发奇想感觉会不会是theme的问题，经过测试果然是，更换其他的theme就能正常显示。仅在win7下测试是这样，其他系统不保证。

收起你的突发奇想，这是luci-app-adguardhome自身的问题，作者rufengsuixing很长时间没有维护了，但是kongfl888
已经二次维护修复了这个问题。
luci-app-adguardhome的问题，就去luci-app-adguardhome的项目下去问。
善用搜索，停止联想。

closed 2020-10-09 by recinfo