coolsnowwolf/lede #5628: LVM2编译出错提示缺少lib

liwenjie119, 2020-10-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：LVM2的版本貌似最近更新到2.03.09了，然后就编译出错提示缺少lib，实际上所需lib已经被勾选

### (2) 路由器型号和固件版本

A：N1

### (3) 详细日志

A：
cp -fpR /home/user/mknop/lede/build_dir/target-aarch64_generic_musl/LVM2.2.03.09/ipkg-install/usr/lib/libdevmapper.so.* /home/user/mknop/lede/build_dir/target-aarch64_generic_musl/LVM2.2.03.09/ipkg-aarch64_generic/libdevmapper/usr/lib
find /home/user/mknop/lede/build_dir/target-aarch64_generic_musl/LVM2.2.03.09/ipkg-aarch64_generic/libdevmapper -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package libdevmapper is missing dependencies for the following libraries:
libselinux.so.1
libsepol.so.1
make[3]: *** [Makefile:115: /home/user/mknop/lede/bin/packages/aarch64_generic/packages/libdevmapper_2.03.09-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/user/mknop/lede/feeds/packages/utils/lvm2'
time: package/feeds/packages/lvm2/compile#60.26#7.00#83.10
make[2]: *** [package/Makefile:113: package/feeds/packages/lvm2/compile] Error 2
make[2]: Leaving directory '/home/user/mknop/lede'
make[1]: *** [package/Makefile:107: /home/user/mknop/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/mknop/lede'
make: *** [/home/user/mknop/lede/include/toplevel.mk:224: world] Error 2

xjqxziii, 2021-02-05:
求问你是怎么解决的? 这个lvm2是哪个包依赖的?

xosadmin, 2021-02-05:
> 求问你是怎么解决的? 这个lvm2是哪个包依赖的?

lvm2是docker依赖的

xjqxziii, 2021-02-05:
> > 求问你是怎么解决的? 这个lvm2是哪个包依赖的?
> 
> lvm2是docker依赖的

这个报错有解吗?

xosadmin, 2021-02-05:
> > > 求问你是怎么解决的? 这个lvm2是哪个包依赖的?
> > 
> > 
> > lvm2是docker依赖的
> 
> 这个报错有解吗?

貌似暂时没有，我尝试过给这个包降级，不会报这个错，但是会出现一些新的问题

closed 2021-06-20 by coolsnowwolf