coolsnowwolf/lede #5630: 内核5.4.68引起的单线复用断网故障

blueveryday, 2020-10-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
最新编译的5.4.68内核固件，上网正常，收视IPTV组播，一切正常，用了半个小时。
然后关掉IPTV播放器，路由器就断网了，wan口显示获取的pppoe拨号信息正常，但是整个网络不能上网了。
log没有错误信息。保持上面1095.783329的内核记录不变。
虽然不能上网了，IPTV可以正常使用。
WAN口跃点是1，MAN口跃点是100
问题原因不明。

目前回退到commit 0ecfa0d4fa08c72595dae904f8f326e5ad94fb89编译的固件，安装上面的操作，使用一切正常。
并且内核日志没有renamed的报错

### (2) 路由器型号和固件版本

A：HC5962 + commit 0ecfa0d4fa08c72595dae904f8f326e5ad94fb89以后的代码编译的固件都有这个问题。

### (3) 详细日志

A：
`[ 16.225452] procd: - init -
[ 16.483968] random: crng init done
[ 16.967217] kmodloader: loading kernel modules from /etc/modules.d/*
[ 17.031080] NET: Registered protocol family 38
[ 17.047622] zram: Added device: zram0
[ 17.057092] ntfs: driver 2.1.32 [Flags: R/O MODULE].
[ 17.112434] cryptodev: driver 1.10 loaded.
[ 17.122225] Loading modules backported from Linux version v5.8-0-gbcf876870b95
[ 17.129463] Backport generated by backports.git v5.8-1-0-g79400d9e
[ 17.158367] usbcore: registered new interface driver ums-alauda
[ 17.165858] usbcore: registered new interface driver ums-cypress
[ 17.173623] usbcore: registered new interface driver ums-datafab
[ 17.181455] usbcore: registered new interface driver ums-freecom
[ 17.189025] usbcore: registered new interface driver ums-isd200
[ 17.196554] usbcore: registered new interface driver ums-jumpshot
[ 17.201452] urngd: v1.0.2 started.
[ 17.204073] usbcore: registered new interface driver ums-karma
[ 17.213764] usbcore: registered new interface driver ums-sddr09
[ 17.221305] usbcore: registered new interface driver ums-sddr55
[ 17.228983] usbcore: registered new interface driver ums-usbat
[ 17.239027] usblp 1-2.3:1.1: usblp0: USB Bidirectional printer dev 3 if 1 alt 0 proto 2 vid 0x03F0 pid 0x3112
[ 17.249358] usbcore: registered new interface driver usblp
[ 17.268093] xt_time: kernel timezone is -0000
[ 17.351373] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=23
[ 17.356835] pci 0000:00:00.0: enabling device (0006 -> 0007)
[ 17.362566] mt7603e 0000:01:00.0: enabling device (0000 -> 0002)
[ 17.368845] mt7603e 0000:01:00.0: ASIC revision: 76030010
[ 17.996026] mt7603e 0000:01:00.0: Firmware Version: ap_pcie
[ 18.001619] mt7603e 0000:01:00.0: Build Time: 20160107100755
[ 18.034666] mt7603e 0000:01:00.0: firmware init done
[ 18.214524] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[ 18.227272] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=24
[ 18.232718] pci 0000:00:01.0: enabling device (0006 -> 0007)
[ 18.238377] mt76x2e 0000:02:00.0: enabling device (0000 -> 0002)
[ 18.244616] mt76x2e 0000:02:00.0: ASIC revision: 76120044
[ 18.647306] mt76x2e 0000:02:00.0: ROM patch build: 20141115060606a
[ 18.657071] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 18.662576] mt76x2e 0000:02:00.0: Build: 1
[ 18.666668] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 18.686678] mt76x2e 0000:02:00.0: Firmware running!
[ 18.693374] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[ 18.732784] PPP generic driver version 2.4.2
[ 18.738395] PPP MPPE Compression module registered
[ 18.744348] NET: Registered protocol family 24
[ 18.759357] kmodloader: done loading kernel modules from /etc/modules.d/*
[ 22.257590] zram0: detected capacity change from 0 to 127926272
[ 22.841735] EXT4-fs (sda1): recovery complete
[ 22.852173] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts:
[ 23.086697] Adding 524284k swap on /mnt/sda1/opt/.swap. Priority:-2 extents:5 across:1040380k
[ 27.368977] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[ 27.381988] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[ 27.390425] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 27.392838] device eth0 entered promiscuous mode
[ 27.404434] mt7530 mdio-bus:1f lan1: configuring for phy/gmii link mode
[ 27.411668] 8021q: adding VLAN 0 to HW filter on device lan1
[ 27.420933] br-lan: port 1(lan1) entered blocking state
[ 27.426310] br-lan: port 1(lan1) entered disabled state
[ 27.432525] device lan1 entered promiscuous mode
[ 27.457416] mt7530 mdio-bus:1f lan2: configuring for phy/gmii link mode
[ 27.465575] 8021q: adding VLAN 0 to HW filter on device lan2
[ 27.475750] br-lan: port 2(lan2) entered blocking state
[ 27.481086] br-lan: port 2(lan2) entered disabled state
[ 27.487814] device lan2 entered promiscuous mode
[ 27.499886] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[ 27.507269] 8021q: adding VLAN 0 to HW filter on device lan3
[ 27.516409] br-lan: port 3(lan3) entered blocking state
[ 27.521697] br-lan: port 3(lan3) entered disabled state
[ 27.528409] device lan3 entered promiscuous mode
[ 27.539475] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 27.547420] 8021q: adding VLAN 0 to HW filter on device wan
[ 29.595261] mt7530 mdio-bus:1f wan: Link is Up - 100Mbps/Full - flow control off
[ 29.602755] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[ 29.865009] br-lan: port 4(wlan0) entered blocking state
[ 29.870487] br-lan: port 4(wlan0) entered disabled state
[ 29.876634] device wlan0 entered promiscuous mode
[ 29.881804] br-lan: port 4(wlan0) entered blocking state
[ 29.887198] br-lan: port 4(wlan0) entered forwarding state
[ 29.893112] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[ 30.403285] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 30.874401] br-lan: port 5(wlan1) entered blocking state
[ 30.879883] br-lan: port 5(wlan1) entered disabled state
[ 30.885924] device wlan1 entered promiscuous mode
[ 31.925760] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[ 31.932411] br-lan: port 5(wlan1) entered blocking state
[ 31.937795] br-lan: port 5(wlan1) entered forwarding state
[ 50.873061] pppoe-wan: renamed from ppp0
[ 541.156600] pppoe-wan: renamed from ppp0
[ 936.985130] pppoe-wan: renamed from ppp0
[ 1073.141372] pppoe-wan: renamed from ppp0
[ 1095.783329] pppoe-wan: renamed from ppp0`

没有问题的固件，最后几排不会有pppoe-wan: renamed from ppp0的报错，其他内核log都一样。

Myoko, 2020-10-09:
试试！4.19.138内核

AmadeusGhost, 2020-10-09:
> 内核5.4.68引起的单线复用断网故障

内核 5.4：太难了，出事了都是我背锅
> 试试 4.19.138内核

他是 ramips，哪来的 4.19 内核。。

大概看了下可能是官方部分 commit 的问题，有空的时候我整理下看看 (((

Myoko, 2020-10-09:
> > 内核5.4.68引起的单线复用断网故障
> 
> 内核 5.4：太难了，出事了都是我背锅
> 
> > 试试 4.19.138内核
> 
> 他是 ramips，哪来的 4.19 内核。。
> 
> 大概看了下可能是官方部分 commit 的问题，有空的时候我整理下看看 (((

刚才稍微看了一下！第一次没注意看是ramips

blueveryday, 2020-10-09:
> > 内核5.4.68引起的单线复用断网故障
> 
> 内核 5.4：太难了，出事了都是我背锅
> 
> > 试试 4.19.138内核
> 
> 他是 ramips，哪来的 4.19 内核。。
> 
> 大概看了下可能是官方部分 commit 的问题，有空的时候我整理下看看 (((

谢谢你。
新代码真心有问题哈。
5.4那个commit就正常使用。

AmadeusGhost, 2020-10-09:
试试这个 commit : https://github.com/AmadeusGhost/lede/commit/7a49d2cf99bd59506bbd9239e0bde81a61f93c40
```
wget https://github.com/AmadeusGhost/lede/commit/7a49d2cf99bd59506bbd9239e0bde81a61f93c40.patch
git apply 7a49d2cf99bd59506bbd9239e0bde81a61f93c40.patch
```

blueveryday, 2020-10-09:
> 试试这个 commit : [AmadeusGhost@7a49d2c](https://github.com/AmadeusGhost/lede/commit/7a49d2cf99bd59506bbd9239e0bde81a61f93c40)
> 
> ```
> wget https://github.com/AmadeusGhost/lede/commit/7a49d2cf99bd59506bbd9239e0bde81a61f93c40.patch
> git apply 7a49d2cf99bd59506bbd9239e0bde81a61f93c40.patch
> ```

感谢你的patch，不好意思今天在外出差，下个月初才能回家测试了。

blueveryday, 2020-10-09:
> 试试这个 commit : [AmadeusGhost@7a49d2c](https://github.com/AmadeusGhost/lede/commit/7a49d2cf99bd59506bbd9239e0bde81a61f93c40)
> 
> ```
> wget https://github.com/AmadeusGhost/lede/commit/7a49d2cf99bd59506bbd9239e0bde81a61f93c40.patch
> git apply 7a49d2cf99bd59506bbd9239e0bde81a61f93c40.patch
> ```

编译固件时报错：

> 
ago@h-PC:~/Desktop/lede$ make -j1 V=s
make[1]: Entering directory '/home/ago/Desktop/lede'                                                                                                                                                                 
make[2]: Entering directory '/home/ago/Desktop/lede'                                                                                                                                                                 
make[3]: Entering directory '/home/ago/Desktop/lede/target/linux'                                                                                                                                                    
make[4]: Entering directory '/home/ago/Desktop/lede/target/linux/ramips'                                                                                                                                             
rm -rf /home/ago/Desktop/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621                                                                                                                                  
mkdir -p /home/ago/Desktop/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621                                                                                                                                
xzcat /home/ago/Desktop/lede/dl/linux-5.4.68.tar.xz | tar -C /home/ago/Desktop/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621 -xf -                                                                      
cp -fpR "/home/ago/Desktop/lede/target/linux/generic/files"/. "/home/ago/Desktop/lede/target/linux/ramips/files"/. /home/ago/Desktop/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.68/        
find /home/ago/Desktop/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.68/ -name \*.rej -or -name \*.orig | xargs -r rm -f                                                                      
if [ -d /home/ago/Desktop/lede/target/linux/generic/patches ]; then echo "generic patches directory is present. please move your patches to the pending directory" ; exit 1; fi                                      
                                                                                                                                                                                                                     
Applying /home/ago/Desktop/lede/target/linux/generic/backport-5.4/010-Kbuild-don-t-hardcode-path-to-awk-in-scripts-ld-vers.patch using plaintext:                                                                    
patching file scripts/ld-version.sh                                                                                                                                                                                  
                                                                                                                                                                                                                     
Applying /home/ago/Desktop/lede/target/linux/generic/backport-5.4/011-kbuild-export-SUBARCH.patch using plaintext:                                                                                                   
patching file Makefile                                                                                                                                                                                               
                                                                                                                                                                                                                     
Applying /home/ago/Desktop/lede/target/linux/generic/backport-5.4/020-backport_netfilter_rtcache.patch using plaintext:                                                                                              
patching file include/net/netfilter/nf_conntrack_extend.h                                                                                                                                                            
patching file include/net/netfilter/nf_conntrack_rtcache.h                                                                                                                                                           
patching file net/netfilter/Kconfig                                                                                                                                                                                  
patching file net/netfilter/Makefile                                                                                                                                                                                 
patching file net/netfilter/nf_conntrack_rtcache.c                                                                                                                                                                   
                                                                                                                                                                                                                     
[... cut ...]
make[3]: *** [Makefile:13: compile] Error 2
make[3]: Leaving directory '/home/ago/Desktop/lede/target/linux'
time: target/linux/compile#14.94#8.58#24.19
make[2]: *** [target/Makefile:25: target/linux/compile] Error 2
make[2]: Leaving directory '/home/ago/Desktop/lede'
make[1]: *** [target/Makefile:18: /home/ago/Desktop/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ago/Desktop/lede'
make: *** [/home/ago/Desktop/lede/include/toplevel.mk:224：world] 错误 2

>

xylz0928, 2020-10-10:
同样的现象，mips架构 7621，
同样情况下单线复用出现故障，wan口频繁断线，目前回退到了上周末了

blueveryday, 2020-10-10:
> 同样的现象，mips架构 7621，
> 同样情况下单线复用出现故障，wan口频繁断线，目前回退到了上周末了

我不在家，没法编译测试，你能不能打一下上面大神的patch。

bunimeidefan, 2020-10-10:
我编译的小米AC2100MT7621的固件，内核日志也是这个报错，WAN口lan口都获得IP跟DNS了，就是上不了网

blueveryday, 2020-10-10:
> 我编译的小米AC2100MT7621的固件，内核日志也是这个报错，WAN口lan口都获得IP跟DNS了，就是上不了网

我不在家，无法测试。
你按照上面大神的patch，重新编译测试一下呢。

xylz0928, 2020-10-11:
> > 同样的现象，mips架构 7621，
> > 同样情况下单线复用出现故障，wan口频繁断线，目前回退到了上周末了
> 
> 我不在家，没法编译测试，你能不能打一下上面大神的patch。

> > 同样的现象，mips架构 7621，
> > 同样情况下单线复用出现故障，wan口频繁断线，目前回退到了上周末了
> 
> 我不在家，没法编译测试，你能不能打一下上面大神的patch。

刚刚使用这个patch在5.4.69内核上编译没有再出现问题，同时感谢一下上面的大神 @AmadeusGhost

blueveryday, 2020-10-11:
> > > 同样的现象，mips架构 7621，
> > > 同样情况下单线复用出现故障，wan口频繁断线，目前回退到了上周末了
> > 
> > 
> > 我不在家，没法编译测试，你能不能打一下上面大神的patch。
> 
> > > 同样的现象，mips架构 7621，
> > > 同样情况下单线复用出现故障，wan口频繁断线，目前回退到了上周末了
> > 
> > 
> > 我不在家，没法编译测试，你能不能打一下上面大神的patch。
> 
> 刚刚使用这个patch在5.4.69内核上编译没有再出现问题，同时感谢一下上面的大神 @AmadeusGhost

好消息！

PR #5650 (2020-10-12): ramips: fix pppoe can't be used under single-line multiplexing
fix #5630
closed 2020-10-12 by coolsnowwolf