coolsnowwolf/lede #5636: 编译luci-app-filetransfer时遇到错误，望各位大神出个招，感激不尽

trombonist852, 2020-10-09:
最近更新到最新源码后编译luci-app-filetransfer时会跳出
wangjun@Trb:~/openwrt$ make package/custom/luci-app-filetransfer/{clean,compile} V=s
time: target/linux/prereq#0.06#0.01#0.08
make[1]: Entering directory '/home/wangjun/openwrt'
make[2]: Entering directory '/home/wangjun/openwrt/package/custom/luci-app-filetransfer'
/home/wangjun/openwrt/feeds/luci/luci.mk:305: *** Package/luci-i18n-filetransfer-zh-cn is missing the VERSION field.  Stop.
make[2]: Leaving directory '/home/wangjun/openwrt/package/custom/luci-app-filetransfer'
time: package/custom/luci-app-filetransfer/clean#0.80#0.02#0.82
package/Makefile:111: recipe for target 'package/custom/luci-app-filetransfer/clean' failed
make[1]: *** [package/custom/luci-app-filetransfer/clean] Error 2
make[1]: Leaving directory '/home/wangjun/openwrt'
/home/wangjun/openwrt/include/toplevel.mk:234: recipe for target 'package/custom/luci-app-filetransfer/clean' failed
make: *** [package/custom/luci-app-filetransfer/clean] Error 2

这个问题去谷歌了一圈也没搜出个所以然，望大神们出个招，这个是啥问题导致的？？感激不尽！

coolsnowwolf, 2020-10-10:
po 语言包出错

AmadeusGhost, 2020-10-11:
这个是官方 master luci 的问题，照着这个 commit 改下：https://github.com/project-openwrt/luci/commit/182725c0f65b27a358c9cbad80a55997f4f16f00

trombonist852, 2020-10-11:
> po 语言包出错

> 这个是官方 master luci 的问题，照着这个 commit 改下：[project-openwrt/luci@182725c](https://github.com/project-openwrt/luci/commit/182725c0f65b27a358c9cbad80a55997f4f16f00)

感谢两位大神！！按照commit已编译成功！

AmadeusGhost, 2020-10-13:
This issue can be closed?

closed 2020-10-13 by trombonist852

trombonist852, 2020-10-13:
关闭啦，谢谢大神！！