coolsnowwolf/lede #5638: 请教一个DNS问题

maodou38, 2020-10-10:
使用shadowsocksR plus，但是就是无法访问raw.githubXXXXXXXX域名，chrome显示是DNS解析不出来，此时电脑NDS用的是软路由地址。
在软路由DHCP中强制绑定1.1.1.1 dns给电脑后，可以正常访问了，但是此时访问很多国内网站就会很卡。
软路由上直接ping raw.githubXXXXXXXXXXX是可以直接ping通的，但是为什么软路由作为电脑的DNS时就无法访问通了呢？

maodou38, 2020-10-10:
找到了问题所在，是我的电脑 win10 虽然DNS里有了软路由地址，但是默认走的却是ipv6本地DNS，导致google啥的都有解析，但是raw.githubXXXX就没有了解析。目前临时方案我是把网卡的ipv6关闭了，希望告知路由器需要怎么配置，才能解决ipv6问题

minlang112, 2020-10-10:
> 找到了问题所在，是我的电脑 win10 虽然DNS里有了软路由地址，但是默认走的却是ipv6本地DNS，导致google啥的都有解析，但是raw.githubXXXX就没有了解析。目前临时方案我是把网卡的ipv6关闭了，希望告知路由器需要怎么配置，才能解决ipv6问题

强制这个域名走代理

maodou38, 2020-10-11:
> > 找到了问题所在，是我的电脑 win10 虽然DNS里有了软路由地址，但是默认走的却是ipv6本地DNS，导致google啥的都有解析，但是raw.githubXXXX就没有了解析。目前临时方案我是把网卡的ipv6关闭了，希望告知路由器需要怎么配置，才能解决ipv6问题
> 
> 
> 
> 强制这个域名走代理

和这没有关系，任何网站都得先过dns才能访问。现在的问题是win10的本地路由有这个域名解析记录，但是是0.0.0.0
所以不会走软路由解析

Ljzkirito, 2020-10-13:
你的意思是ipv6的dns不是指向路由吗？
如果你删除了IPv6 ULA 前缀，尝试加回去，看看ipv6的dns是不是路由的ipv6，能不能解决问题

maodou38, 2020-10-13:
不是 路由别的都没问题 但是GitHub的raw存储服务器上不去 ipv6那个dns出来的是0.0.0.0 所以不会再走ipv4路由

Ljzkirito, 2020-10-13:
[image]
我这里很正常，要不重置路由试试？

closed 2021-06-20 by coolsnowwolf