coolsnowwolf/lede #5643: 使用 lede 源码编译 R2S ，不开机

chpos, 2020-10-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用 lede 源码给 R2S 编译 固件，make menuconfig 选择步骤是  rockchip ——rockchip_armv8 ——friendlyarm_nanopi-r2s  

然后步骤是make -j8 download V=s  -----make -j1 V=s ，操作步骤完全按照 lean 的指示来的， 固件编译成功，多次刷机无法进入系统，R2S 灯不亮，换别人的固件就可以

### (2) 路由器型号和固件版本

A：R2S  rk3328

### (3) 详细日志

A：R2S 灯不亮  无任何反应

asushugo, 2020-10-11:
#5614 这里也有遇到这个问题的，可能是最近这两次rockchip修改的commit导致的，现在我回滚之前的代码继续用了

chpos, 2020-10-11:
> #5614 这里也有遇到这个问题的，可能是最近这两次rockchip修改的commit导致的，现在我回滚之前的代码继续用了

回滚代码怎么回滚，如果是 rockchip修改的commit 导致的问题，不知啥时候能修好

varios009, 2020-10-11:
也有遇到这个问题~
最奇怪的是我用了全金属外壳，只要lede 源码编译 R2S的 固件。就开不了机。
只要拆了全金属外壳,就可以开机。(99.9%排除了主板和全金属外壳短路问题，主板都贴上的胶布，主板四边都用了有机硅密封胶)
换别人的固件刷在同一SD卡上就可以正常开机。不理有没有装全金属外壳.

whalechoi, 2020-10-11:
> 也有遇到这个问题~
> 最奇怪的是我用了全金属外壳，只要lede 源码编译 R2S的 固件。就开不了机。
> 只要拆了全金属外壳,就可以开机。(99.9%排除了主板和全金属外壳短路问题，主板都贴上的胶布，主板四边都用了有机硅密封胶)
> 换别人的固件刷在同一SD卡上就可以正常开机。不理有没有装全金属外壳.

啊这。。。我试了下确实可以。
难道这就是传说中的拆壳开机法？

varios009, 2020-10-11:
> > 也有遇到这个问题~
> > 最奇怪的是我用了全金属外壳，只要lede 源码编译 R2S的 固件。就开不了机。
> > 只要拆了全金属外壳,就可以开机。(99.9%排除了主板和全金属外壳短路问题，主板都贴上的胶布，主板四边都用了有机硅密封胶)
> > 换别人的固件刷在同一SD卡上就可以正常开机。不理有没有装全金属外壳.
> 
> 啊这。。。我试了下确实可以。
> 难道这就是传说中的拆壳开机法？
> 内存卡用的四年前买的32G金士顿C10U3红卡。

已经试了东芝32G,创见64G, 三星32G,夏科32G.只要装全金属外壳，刷lede 源码编译 R2S的 固件,只有夏科32G,两三次可以开一次机。这4种SD卡刷其它都可以正常开机。刷lede 源码编译 R2S的 固件,经过多次测试只全金属外壳装不要太紧，上下壳空一条裂缝出来就能开机。(已经排除不是太大力压到CPU的问题)。只能等真大神出来解释。。。。。

whalechoi, 2020-10-12:
> 已经试了东芝32G,创见64G, 三星32G,夏科32G.只要装全金属外壳，刷lede 源码编译 R2S的 固件,只有夏科32G,两三次可以开一次机。这4种SD卡刷其它都可以正常开机。刷lede 源码编译 R2S的 固件,经过多次测试只全金属外壳装不要太紧，上下壳空一条裂缝出来就能开机。(已经排除不是太大力压到CPU的问题)。只能等真大神出来解释。。。。。

#5596 这个里面有个关于内存卡的补丁，去掉就好了

101-rockchip-rk3328-fix-nanopi-r2s-failed-to-boot-on-som.patch

chpos, 2020-10-12:
> #5596 这个里面有个关于内存卡的补丁，去掉就好了
>
> 101-rockchip-rk3328-fix-nanopi-r2s-failed-to-boot-on-som.patch

这个 内存卡的补丁在哪个文件位置去除，大佬

6667933, 2020-10-12:
> 这个 内存卡的补丁在哪个文件位置去除，大佬

target/linux/rockchip/patches-5.4

varios009, 2020-10-12:
> 这个 内存卡的补丁在哪个文件位置去除，大佬

101-rockchip-rk3328-fix-nanopi-r2s-failed-to-boot-on-som.patch
新的已经没有这个文件了~但编译出来的固件一样我有上面说的情况~

6667933, 2020-10-12:
boot/uboot-rockchip/patches/101-rockchip-rk3328-fix-nanopi-r2s-failed-to-boot-on-som.patch

6667933, 2020-10-12:
我这边删除packege/boot/uboot-rockchip/patches/101-rockchip-rk3328-fix-nanopi-r2s-failed-to-boot-on-som.patch

后重新编译也一样症状

varios009, 2020-10-12:
> 我这边删除packege/boot/uboot-rockchip/patches/101-rockchip-rk3328-fix-nanopi-r2s-failed-to-boot-on-som.patch
> 
> 后重新编译也一样症状

对的~我也试了删除那个文件 后重新编译也一样症状

6667933, 2020-10-12:
对比了最近的PR，发现是温度显示的锅，删除这个就行了。target/linux/rockchip/armv8/base-files/lib/preinit/82_show_temp #5656

467815891a, 2020-10-12:
> 对比了最近的PR，发现是温度显示的锅，删除这个就行了。target/linux/rockchip/armv8/base-files/lib/preinit/82_show_temp #5656

不好意思，是我提交的，在我这里没有问题，我就提交了。没想到出了问题

closed 2020-10-12 by coolsnowwolf

chpos, 2020-10-13:
我最近尝试了一下问题已经解决，问题和101文件没有关系，是温度显示的问题，去掉 target/linux/rockchip/armv8/base-files/lib/preinit/82_show_temp 这个文件就就可以了

PR #5656 (2020-10-12): remove 82_show_temp to resolve r2s start up problems
Q：你知道这是`pull request`吗？(使用 "x" 选择)
* [x] 我知道
 revert #5612  to resolve nanopi r2s startup problems,fix #5643