coolsnowwolf/lede #5644: 编译出错，希望大佬帮忙看看

crazydayli, 2020-10-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：/bin/sh: 1: ../../third_party/llvm-build/Release+Asserts/bin/clang++: not found
ninja: build stopped: subcommand failed.
Makefile:83: recipe for target '/home/jin/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-4/.built' failed
make[3]: *** [/home/jin/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-4/.built] Error 1
make[3]: Leaving directory '/home/jin/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#10.29#5.84#10.29
package/Makefile:111: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[2]: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/jin/lede'
package/Makefile:107: recipe for target '/home/jin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/jin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jin/lede'
/home/jin/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
看着像naiveproxy出问题，但是在ssrp里没有勾选编译，也重新下载过包了

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：/bin/sh: 1: ../../third_party/llvm-build/Release+Asserts/bin/clang++: not found
ninja: build stopped: subcommand failed.
Makefile:83: recipe for target '/home/jin/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-4/.built' failed
make[3]: *** [/home/jin/lede/build_dir/target-x86_64_musl/naiveproxy-85.0.4183.83-4/.built] Error 1
make[3]: Leaving directory '/home/jin/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#10.29#5.84#10.29
package/Makefile:111: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[2]: *** [package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/jin/lede'
package/Makefile:107: recipe for target '/home/jin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/jin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jin/lede'
/home/jin/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

mulder125, 2020-10-11:
你應該有加入另一位L大老的源?
應該是下面的問題，待修正
https://github.com/xiaorouji/openwrt-package/commit/35e58aa8104292b93a82072d955373d11de1de4e#comments

修正前可以先取消SSRP 和 passwall的naiveproxy支持，並將Network下的naiveproxy取消勾選

crazydayli, 2020-10-11:
不勾选naiveproxy编译了一遍，清空，勾选了这个编译错误代码一样，求教

crazydayli, 2020-10-11:
> 你應該有加入另一位L大老的源?
> 應該是下面的問題，待修正
> [xiaorouji/openwrt-package@35e58aa#comments](https://github.com/xiaorouji/openwrt-package/commit/35e58aa8104292b93a82072d955373d11de1de4e#comments)
> 
> 修正前可以先取消SSRP 和 passwall的naiveproxy支持，並將Network下的naiveproxy取消勾選

马上回去试一试，之前2个都没有勾选，还没看network里

crazydayli, 2020-10-11:
成功编译感谢

closed 2020-10-11 by crazydayli