coolsnowwolf/lede #5646: moonlight串流被中断

yaya5508190, 2020-10-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)该版本的固件配置端口转发后，moonlight外网串流连接上后可以显示画面，但很快就被中断, OpenWrt R9.12.31 / LuCI Master (git-19.338.43082-9e87e69)版本的固件就没有该问题

OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)会中断的固件版本抓包出现大量ICMP Destination unreachable (Port unreachable)
[image]
出现大量ICMP Destination unreachable (Port unreachable)
OpenWrt R9.12.31 / LuCI Master (git-19.338.43082-9e87e69)正常不会被中断的抓包没有ICMP的请求
[image]

希望有大神可以帮忙看看是什么原因

### (2) 路由器型号和固件版本

A：X86 Exsi

### (3) 详细日志

A：两个固件版本的防火墙配置一致 防火墙配置如下
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option fullcone '1'
	option flow_offloading '1'
	option flow_offloading_hw '0'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	option network 'wan wan6'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config include
	option path '/etc/firewall.user'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '5800'
	option dest_ip '172.16.188.220'
	option dest_port '3389'
	option name 'mstsc'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '47984-48010'
	option dest_port '47984-48010'
	option name 'game-stream'
	option dest_ip '172.16.188.164'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '7801'
	option dest_ip '172.16.188.100'
	option dest_port '9000'
	option name 'portainer'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '7800'
	option dest_ip '172.16.188.100'
	option name 'centos7'
	option dest_port '22'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '9295-9304'
	option dest_ip '172.16.188.191'
	option dest_port '9295-9304'
	option name 'ps4'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '987'
	option dest_ip '172.16.188.191'
	option dest_port '987'
	option name 'ps4'

config redirect
	option target 'DNAT'
	option src 'wan'
	option dest 'lan'
	option proto 'tcp udp'
	option src_dport '4443'
	option dest_ip '172.16.188.100'
	option dest_port '4443'
	option name 'open-connect'

config include 'zerotier'
	option type 'script'
	option path '/etc/zerotier.start'
	option reload '1'

config include 'miniupnpd'
	option type 'script'
	option path '/usr/share/miniupnpd/firewall.include'
	option family 'any'
	option reload '1'

config include 'adbyby'
	option type 'script'
	option path '/usr/share/adbyby/firewall.include'
	option reload '1'

config rule 'adblock'
	option name 'adblock'
	option target 'DROP'
	option src 'wan'
	option proto 'tcp'
	option dest_port '8118'

config include 'ipsecd'
	option type 'script'
	option path '/etc/ipsec.include'
	option reload '1'

config rule 'ike'
	option name 'ike'
	option target 'ACCEPT'
	option src 'wan'
	option proto 'udp'
	option dest_port '500'

config rule 'ipsec'
	option name 'ipsec'
	option target 'ACCEPT'
	option src 'wan'
	option proto 'udp'
	option dest_port '4500'

config rule 'ah'
	option name 'ah'
	option target 'ACCEPT'
	option src 'wan'
	option proto 'ah'

config rule 'esp'
	option name 'esp'
	option target 'ACCEPT'
	option src 'wan'
	option proto 'esp'

config rule 'kms'
	option name 'kms'
	option target 'ACCEPT'
	option src 'wan'
	option proto 'tcp'
	option dest_port '1688'

config include 'mia'
	option type 'script'
	option path '/etc/mia.include'
	option reload '1'

config rule 'gre'
	option name 'gre'
	option target 'ACCEPT'
	option src 'wan'
	option proto '47'

yaya5508190, 2020-10-17:
排查后发现应该是端口转发未完全生效的问题，具体为什么未完全生效 原因就不清楚了 自己手动添加防火墙规则

closed 2020-10-17 by yaya5508190

Jecvay, 2021-02-21:
@yaya5508190 请问有没有解决，我碰到一样的问题

hyne-shan, 2021-07-18:
我也遇到一样的问题 要哭了

MikuMikuyukis, 2022-06-22:
turbo acc 里面关闭sfe加速即可解决，亲测有效

CrazyBunQnQ, 2023-08-02:
> turbo acc 里面关闭sfe加速即可解决，亲测有效

可算找到解决办法了！

jamestang0219, 2024-02-06:
> turbo acc 里面关闭sfe加速即可解决，亲测有效

厉害了！关掉软件流量分载和SFE之后，果然就可以在公网用moonlight了！