coolsnowwolf/lede #5649: 实时流量统计不准确，有没有更准确点的流量统计插件

kleisth, 2020-10-11:
## 1.实时流量监测插件wrtbwmon不准确

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
固件自带的wrtbwmon实时流量监测不准确，即使客户机速度跑到了120MB/s，监测这里还是显示几KB/s甚至几b/s。
是否设置不当或者有其他更为准确的插件，求告知，万分感谢！

A：

### (2) 路由器型号和固件版本
x86_64软路由，最新的源码编译

A：

### (3) 详细日志
无特别日志。

A：

sxx1314, 2020-10-12:
请关闭 sfe等加速

kleisth, 2020-10-12:
> 请关闭 sfe等加速
非常感谢！
果然可以了，这是啥原理？sfe会影响流量监测的吗？说明只写会影响多wan和qos - -b。。。
Flow Offload呢？也会影响吗？
我cpu够强，所以开不开这类软加速倒是没啥影响。只是好奇想了解下。。。

sxx1314, 2020-10-13:
> > 请关闭 sfe等加速
> > 非常感谢！
> > 果然可以了，这是啥原理？sfe会影响流量监测的吗？说明只写会影响多wan和qos - -b。。。
> > Flow Offload呢？也会影响吗？
> > 我cpu够强，所以开不开这类软加速倒是没啥影响。只是好奇想了解下。。。

flow offload 一样会影响  任何硬加速如hwnat及软加速都会影响流量监控。具体你可以自行了解sfe等加速原理。

Dimitri2020007, 2020-10-13:
luci的netdata和vnstat不准确吗？

closed 2021-06-20 by coolsnowwolf