coolsnowwolf/lede #5653: 请教两个关于luci界面的问题

recinfo, 2020-10-12:
1、现在编译成功系统中，菜单内的插件排序不是很满意，想修改一下，比如现在的排序是:
**服务
    A
    B
    C**
想修改成
**服务
    A
   C
    B**
那么我应该修改哪个文件，如果是想在编译前就修改好，应该修改哪个文件？？
2、如何修改 **Powered by LuCI Master (git-20.256.12360-1a54222) / OpenWrt R20.9.15** 内容？在编译前能修改么，应该修改哪个文件？

lphgor, 2020-10-12:
1、顺序是可以设定的，在每个app的controller文件里面定义，`entry(path, target, title=nil, order=nil)`，最后一个参数就是顺序，数字越小越靠前。
例如：kodexplorer 是 44
https://github.com/coolsnowwolf/lede/blob/289296beec16e7eb9f9eb64a6b7d24145064beb7/package/lean/luci-app-kodexplorer/luasrc/controller/kodexplorer.lua#L10

2、“Powered by”在每个主题的`footer.htm`文件里面https://github.com/coolsnowwolf/luci/blob/1a54222c61456586c31457fda62ac01034f1ab23/themes/luci-theme-bootstrap/luasrc/view/themes/bootstrap/footer.htm#L17

recinfo, 2020-10-12:
明白了，多谢指教。

closed 2020-10-12 by recinfo