coolsnowwolf/lede #5654: 最新源码编译好ddns-scripts_aliyun设置好后提示错误，望大神们帮忙看看，感激不尽

trombonist852, 2020-10-12:
设置好后提示
151700       : verbose mode  : 0 - run normal, NO console output
 151700  WARN : No update_url found/defined or no update_script found/defined! - TERMINATE
 151700  WARN : PID '16953' exit WITH ERROR '1' at 2020-10-12 15:17
这个是啥原因？之前的版本已知运行的很好，新源码编译好之后刚开始连阿里云都不显示，后来在etc/config里面建立一个ddns的文件夹，阿里云就可以显示了，但是设置好会提示这个错误

sola97, 2021-01-04:
相同问题

LIU-LIU-LIU, 2021-05-25:
同问，这个问题

closed 2021-06-20 by coolsnowwolf

k7002, 2022-01-22:
或许可以参考这个https://forum.openwrt.org/t/solved-netgear-r7800-ddns-script-terminates-with-error/13345

cyqw, 2022-07-17:
界面配置可以使用custom的provider
看了原始版本的/usr/lib/ddns/dynamic_dns_updater.sh文件，这个文件不支持custom的provider。

cyqw, 2022-07-17:
界面配置可以使用custom的provider
看了原始版本的/usr/lib/ddns/dynamic_dns_updater.sh文件，这个文件不支持custom的provider。
发现还有很多提供商的地址可以单独安装的。