coolsnowwolf/lede #5660: R2S 怎么编译出能显示频率和温度的固件？

chpos, 2020-10-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：昨天使用lean 的源码编译R2S 固件时不开机的情况已经解决了，今天成功的编译成功固件并且可以开机，但是我自己编译的固件和别人的固件还不一样，在主页的 状态 ———概览里面，别人固件的主机型号显示的是：FriendlyElec NanoPi R2S（CpuMark:19924.453115 Scores)，我的固件主机型号显示是：FriendlyElec NanoPi R2S，没有 CpuMack
别人固件的架构显示的是 ARMv8 Processor x4 (1200MHz, 49.1 ℃），我的固件架构显示为： ARMv8 Processor x4，不显示频率和温度， 而且别人的固件在 系统 选项里面有一个 CPU 性能优化调节，而我的没有，我查了一下谷歌，发现这方面知识很少，其中有一个帖子是这样说的 ：在 lede 目录下 添加 enable_autocore.diff 文件，我试了一下，直接编译报错！
https://github.com/coolsnowwolf/lede/issues/4072    https://github.com/klever1988/lede/blob/master/enable_autocore.diff
别人编译的固件有的砍掉了某些功能，我自己想要的插件没有，还有的固件是英文系统，所以才自行编译固件，希望有懂的大佬告知我一下

### (2) 路由器型号和固件版本

A：R2S  rk3328

### (3) 详细日志

A：主页不显示 CpuMark，CPU 频率和温度，系统选项里面没有 CPU 性能优化调节

asushugo, 2020-10-13:
看刚刚的提交，应该是勾上autocore-arm就行了

chpos, 2020-10-13:
> 看刚刚的提交，应该是勾上autocore-arm就行了

在 luci———Applications 里面选择吗？更新了一下代码，没看到

AmadeusGhost, 2020-10-13:
在 extra package 选择

chpos, 2020-10-13:
找到了，点开后 发现 autocore-arm  我选择了编译上了，应该不是这个选择项

asushugo, 2020-10-13:
你编译一下就知道了，具体你看这个提交 #5657

chpos, 2020-10-13:
> 你编译一下就知道了，具体你看这个提交 #5657

这个提交已经被删除了，这个提交会导致新刷固件不开机

asushugo, 2020-10-13:
> > 你编译一下就知道了，具体你看这个提交 #5657
> 
> 这个提交已经被删除了，这个提交会导致新刷固件不开机

你看清楚，这是新的提交
[image]

chpos, 2020-10-14:
重新编译了一下可以了，这里反应一个问题，编译 R2S 的时候，第一次编译往往会失败，第二次删除配置重新编译就会成功，我遇到两次这样的情况了。感谢 lean 等人的付出，谢谢

AmadeusGhost, 2020-10-23:
问题解决了那关闭下 issue？

closed 2020-10-23 by chpos