coolsnowwolf/lede #5661: dnsmasq logqueries not work

langit7, 2020-10-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
i want to log all dns query to route specific dns ip to specific interface
uci set dhcp.@dnsmasq[0].logdhcp='1'
uci set dhcp.@dnsmasq[0].logqueries='1'
uci commit dhcp
service dnsmasq restart

### (2) 路由器型号和固件版本
Xiaomi Mi Router 3 Pro 
### (3) 详细日志
logread -f  # no dns query displayed

lphgor, 2020-10-13:
改源码自己编译的话，删除这2行：
https://github.com/coolsnowwolf/lede/blob/c39177efcfd9df8ea8e5d8e687d327fe77bfde52/package/lean/default-settings/files/zzz-default-settings#L50-L51

直接改路由器的话，删除`/etc/dnsmasq.conf`里面的这一行：

```
log-facility=/dev/null
```

langit7, 2020-10-13:
Thanks. Work perfect

closed 2020-10-13 by langit7