coolsnowwolf/lede #5663: 单臂路由一直不通

jzhou404, 2020-10-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

单臂路由一直不通。 

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

Ｑ：WAN可以PPPOE拨号成功， ＬＡＮ可以获取ＩＰ，但就是出不去不通，　无法访问外网．老版本4.19内核的固件使用完全正常， 新内核有问题

### (2) 路由器型号和固件版本

A：J3160单口千兆软路由

### (3) 详细日志

A：

gd0772, 2020-10-14:
这不应该是学习编译和反馈Bug的地方么？这么种个人或者编译者的固件问题，也在这里发 issue ？不应该是谁编译的固件找谁么？尊重下开发者吧？使用问题，在群里可以先问问别人，或者自己谷歌关键词，如果是编译或在确实是程序的问题，在来这里发 issue

jzhou404, 2020-10-14:
这是固件编译之后的问题， 不是配置的问题。

同样的编译配置文件(.config)编译出来使用老4.19老内核就没有任何问题。最新的5.4内核上就无法单臂， 显示都正常， 就是不通。

所以， 我反馈一下。

Dimitri2020007, 2020-10-14:
你在dnsmasq上设置了对应的dns转发地址吗？

有没有装了多个dns软件起冲突了无法访问（dnscrypt-proxy、unbound、dnsforwarder）

xwcoco, 2020-10-16:
我也遇到相同的问题了,老内核5.4.61还正常,今天编译一下5.4.69,相同的配置怎么设都连不上网,DNS是正常的,能获取正确的DNS,但就是网页打不开,不知道什么原因.

Lile1, 2020-11-06:
WAN 防火墙没设置

closed 2021-06-20 by coolsnowwolf