coolsnowwolf/lede #5665: 编译naiveproxy错误

jackerjiang, 2020-10-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译naiveproxy报错，不选择naiveproxy编译可以完成，有结局办法吗

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志
ninja: Entering directory `out/Release'
[1/2068] CC obj/base/third_party/xdg_mime/xdg_mime/xdgmimeicon.o
FAILED: obj/base/third_party/xdg_mime/xdg_mime/xdgmimeicon.o 
../../third_party/llvm-build/Release+Asserts/bin/clang -MMD -MF obj/base/third_party/xdg_mime/xdg_mime/xdgmimeicon.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-12-init-3492-ga1caa302-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=cc396d2ee91286924b377c2124b1efbcad7df8f2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/ignores.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-psabi -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-string-concatenation -Wno-max-tokens -std=c11 --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include/nss -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include/nspr -Wno-implicit-fallthrough --target=x86_64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../base/third_party/xdg_mime/xdgmimeicon.c -o obj/base/third_party/xdg_mime/xdg_mime/xdgmimeicon.o
../../third_party/llvm-build/Release+Asserts/bin/clang: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory
[2/2068] CC obj/base/third_party/xdg_mime/xdg_mime/xdgmimemagic.o
FAILED: obj/base/third_party/xdg_mime/xdg_mime/xdgmimemagic.o 
../../third_party/llvm-build/Release+Asserts/bin/clang -MMD -MF obj/base/third_party/xdg_mime/xdg_mime/xdgmimemagic.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-12-init-3492-ga1caa302-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=cc396d2ee91286924b377c2124b1efbcad7df8f2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/ignores.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-psabi -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-string-concatenation -Wno-max-tokens -std=c11 --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include/nss -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include/nspr -Wno-implicit-fallthrough --target=x86_64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../base/third_party/xdg_mime/xdgmimemagic.c -o obj/base/third_party/xdg_mime/xdg_mime/xdgmimemagic.o
../../third_party/llvm-build/Release+Asserts/bin/clang: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory
[3/2068] CXX obj/base/third_party/symbolize/symbolize/symbolize.o
FAILED: obj/base/third_party/symbolize/symbolize/symbolize.o 
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/base/third_party/symbolize/symbolize/symbolize.o.d -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-12-init-3492-ga1caa302-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=cc396d2ee91286924b377c2124b1efbcad7df8f2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/ignores.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-psabi -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-string-concatenation -Wno-max-tokens -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include/nss -I/home/jacker/lede/staging_dir/target-x86_64_musl/usr/include/nspr -fvisibility-inlines-hidden --target=x86_64-openwrt-linux-musl -D_LI [... cut ...]
make[3]: Leaving directory '/home/jacker/lede/feeds/lienol/package/naiveproxy'
time: package/feeds/lienol/naiveproxy/compile#21.93#13.27#16.94
make[2]: *** [package/Makefile:113: package/feeds/lienol/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/jacker/lede'
make[1]: *** [package/Makefile:107: /home/jacker/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jacker/lede'
make: *** [/home/jacker/lede/include/toplevel.mk:224: world] Error 2A：

jackerjiang, 2020-10-14:
已解决

closed 2020-10-14 by jackerjiang

blueveryday, 2020-10-21:
> 已解决

怎么解决的，说说啊。

cpboy2020, 2020-10-29:
怎么解决的，说说啊。我也碰到了

lazymartin, 2020-11-04:
> 怎么解决的，说说啊。我也碰到了

安装依赖 sudo apt-get install -y libncurses5