coolsnowwolf/lede #5672: 关于网页终端ttyd实现方法改进的一点问题，利用nginx反向代理

ZeaKyX, 2020-10-16:
想要实现类似unraid里的网页终端的效果：
打开 `http://ip或域名/webterminal`，就是ttyd，如果当前为未登录状态就跳转至登录页面
unraid中就是使用的ttyd，并且应该是靠nginx实现的。
现在使用默认配置编译的lean op，安装了nginx-mod-luci，安装后自动关闭uhttpd然后掌管luci
然后在`/etc/nginx/conf.d/luci.locations`中加入[ttyd官方配置](https://github.com/tsl0922/ttyd/wiki/Nginx-reverse-proxy)，可以正常代理
**但是未登录状态下可以直接进入**，且路径不在cgi-bin之下，不知道luci是跳转原理究竟是什么，希望有大神赐教

Awrtiger, 2020-10-20:
/etc/config/ttdy 配置文件 添加个验证 增加安全性 就是有点麻烦而已。

`option credential [用户名]:[密码]`

ZeaKyX, 2020-10-20:
> /etc/config/ttdy 配置文件 添加个验证 增加安全性 就是有点麻烦而已。
> 
> `option credential [用户名]:[密码]`

我绕这么一圈就是为了避免这种选择啊，他要是有弹窗登陆也好，浏览器还能自动填

Awrtiger, 2020-10-20:
> > /etc/config/ttdy 配置文件 添加个验证 增加安全性 就是有点麻烦而已。
> > `option credential [用户名]:[密码]`
> 
> 我绕这么一圈就是为了避免这种选择啊，他要是有弹窗登陆也好，浏览器还能自动填

Google浏览器可以自动填充用户名密码

ZeaKyX, 2020-10-20:
> > > /etc/config/ttdy 配置文件 添加个验证 增加安全性 就是有点麻烦而已。
> > > `option credential [用户名]:[密码]`
> > 
> > 
> > 我绕这么一圈就是为了避免这种选择啊，他要是有弹窗登陆也好，浏览器还能自动填
> 
> Google浏览器可以自动填充用户名密码

原来是有弹窗的啊，抱歉孤陋寡闻了。这个要端口转发，占用端口号。不过也算勉强满足需求了。

Dimitri2020007, 2020-10-24:
装了tty的网页版，一片空白是怎么回事啊？

closed 2021-06-20 by coolsnowwolf