coolsnowwolf/lede #5673: HiWiFi HC5962的固件限制在32m，能否扩大一点。

lee29, 2020-10-16:
RT

tcsr200722, 2020-10-17:
楼主，B70在lede下无线信号是不是不如老毛子？

blueveryday, 2020-10-17:
> 楼主，B70在lede下无线信号是不是不如老毛子？

比老毛子好。

programmejava, 2020-10-21:
看这个issue           https://github.com/coolsnowwolf/lede/issues/1547
现在B70好像dts文件名改了，大概位置还在，方法没变，就可以编译超过32M   在系统 ->软件里空间大概100M  但是如果你的极4刷了breed的话，超过32M的固件无法在breed里更新 ，好久没有折腾极4了，不知道现在更新的breed有没有去掉这个限制。

blueveryday, 2020-10-22:
> 看这个issue #1547
> 现在B70好像dts文件名改了，大概位置还在，方法没变，就可以编译超过32M 在系统 ->软件里空间大概100M 但是如果你的极4刷了breed的话，超过32M的固件无法在breed里更新 ，好久没有折腾极4了，不知道现在更新的breed有没有去掉这个限制。

找到方法了。
修改HC5962的dts文件，把label=ubi里面的终止地址改下就可以了。
以下是修改后的参数。

partition@340000 {
label = “ubi”;
reg = <0x340000 0x4d00000>;
};

保存后重新编译即可。

lee29, 2020-10-28:
> > 看这个issue #1547
> > 现在B70好像dts文件名改了，大概位置还在，方法没变，就可以编译超过32M 在系统 ->软件里空间大概100M 但是如果你的极4刷了breed的话，超过32M的固件无法在breed里更新 ，好久没有折腾极4了，不知道现在更新的breed有没有去掉这个限制。
> 
> 找到方法了。
> 修改HC5962的dts文件，把label=ubi里面的终止地址改下就可以了。
> 以下是修改后的参数。
> 
> partition@340000 {
> label = “ubi”;
> reg = <0x340000 0x4d00000>;
> };
> 
> 保存后重新编译即可。

ubi@0 {
				label = "ubi";
				reg = <0x0 0x79c0000>;
			};
是改这段么

programmejava, 2020-10-29:
是的

lee29, 2020-11-03:
> 是的

`/dts-v1/;

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
	compatible = "hiwifi,hc5962", "mediatek,mt7621-soc";
	model = "HiWiFi HC5962";

	aliases {
		led-boot = &led_system;
		led-failsafe = &led_system;
		led-running = &led_status;
		led-upgrade = &led_status;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	leds {
		compatible = "gpio-leds";

		led_status: status {
			label = "hc5962:white:status";
			gpios = <&gpio 6 GPIO_ACTIVE_LOW>;
		};

		led_system: system {
			label = "hc5962:red:system";
			gpios = <&gpio 7 GPIO_ACTIVE_LOW>;
			panic-indicator;
		};
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 18 GPIO_ACTIVE_HIGH>;
			linux,code = <KEY_RESTART>;
		};
	};

	ubi-concat {
		compatible = "mtd-concat";
		devices = <&ubipart0 &ubipart1>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@340000 {
				label = "ubi";
				reg = <0x340000 0x4d00000>;
			};
		};
	};
};

&nand {
	status = "okay";

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		partition@0 {
			label = "u-boot";
			reg = <0x0 0x80000>;
			read-only;
		};

		partition@80000 {
			label = "debug";
			reg = <0x80000 0x80000>;
			read-only;
		};

		factory: partition@100000 {
			label = "factory";
			reg = <0x100000 0x40000>;
			read-only;
		};

		partition@140000 {
			label = "kernel";
			reg = <0x140000 0x400000>;
		};

		ubipart0: partition@540000 {
			label = "ubipart0";
			reg = <0x540000 0x1c80000>;
		};

		partition@21c0000 {
			label = "bdinfo";
			reg = <0x21c0000 0x80000>;
			read-only;
		};

		ubipart1: partition@2240000 {
			label = "ubipart1";
			reg = <0x2240000 0x5d40000>;
		};
	};
};

&pcie {
	status = "okay";
};

&pcie0 {
	mt76@0,0 {
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
		ieee80211-freq-limit = <2400000 2500000>;
	};
};

&pcie1 {
	mt76@0,0 {
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x8000>;
		ieee80211-freq-limit = <5000000 6000000>;
	};
};

&switch0 {
	ports {
		port@1 {
			status = "okay";
			label = "lan1";
		};

		port@2 {
			status = "okay";
			label = "lan2";
		};

		port@3 {
			status = "okay";
			label = "lan3";
		};

		port@4 {
			status = "okay";
			label = "wan";
		};
	};
};

&state_default {
	gpio {
		groups = "uart3", "jtag";
		function = "gpio";
	};
};`

可以编译出超过32m的固件，但在breed中还是有32m限制。是breed的锅吗？有其他的bootloader推荐么

programmejava, 2020-11-04:
可以用pb-boot  也就是潘多拉的那个uboot 地址如下   
http://downloads.pangubox.com:6380/

uoow, 2020-11-10:
> `/dts-v1/;
>
> #include "mt7621.dtsi"
>
> #include <dt-bindings/gpio/gpio.h>
> #include <dt-bindings/input/input.h>
>
> / {
> compatible = "hiwifi,hc5962", "mediatek,mt7621-soc";
> model = "HiWiFi HC5962";
>
> ``` [...]

固件编译出来的大小与此文件无关吧。
只是修改了此文件后编译出来的固件刷入后还是不成功，估计是修改得不对，这个修改方法（以前的ISSSUE里）是很久之前编译固件有效，现在不行。需要大神们再次修改适配

uoow, 2020-11-10:
> 
> 
> 可以用pb-boot 也就是潘多拉的那个uboot 地址如下
> http://downloads.pangubox.com:6380/
> 
> 或者刷过sysupgrade固件，在openwrt中升级，这个我也没试过 有时间可以折腾一下。

翻issues才发现之前解决的是此大神，希望早日折腾出来。
我的就是pb-boot，最近编译的固件怎么刷都不成功，编译没问题，感觉就是固件大小影响了

closed 2021-06-20 by coolsnowwolf