coolsnowwolf/lede #5678: R2S 不稳定

hxlls, 2020-10-18:
这几个月一直在测试稳定性，发现好像一直不能稳定运行，通过对比其他稳定的固件，最后锁定了好像大雕的源码的超频到了1608MHZ ，在这个频率下，有概率引起在高负荷运行时候会重启或者掉网卡等故障，对比的固件是https://github.com/klever1988/nanopi-openwrt ，这个能持续稳定运行，并且反映稳定性也不错，固件的上限频率只到了1512，因为随机重启或者掉网卡后没法查看日志，所以也没法确认到底是超频引起供电不足或者还是机器的体制问题？
需求：怎么更改超频的频率上限？或者大佬能把频率上限下调点？

hxlls, 2020-10-18:
找到解决办法了，https://github.com/armbian/build/raw/master/patch/kernel/rockchip64-dev/RK3328-enable-1512mhz-opp.patch
把上面链接内容替换掉/home/ling/lede/target/linux/rockchip/patches-5.4/003-arm64-dts-rockchip-add-more-cpu-operating-points-for.patch里面的内容，就会上限是1512MHZ 实际测试10次coremark 在没替换前有概率60%会重启，替换后10次稳过,目测是供电问题，更换小点的电源，也能跑过，好像对TF卡兼容性略好点？其中有张TF卡，以前启动不了，现在能启动了。有类似的问题可以尝试这种方式解决，不过解决之前，记得先更换好的电源试试.  PS：友善的这个产品设计。。有点太坑了，5V3A 的microUSB电源还不好买。。

hxlls, 2020-10-18:
实际测试截图
[image]

asushugo, 2020-10-19:
我也有试过第一次启动后，过一段时间会有一定的概率重启，之后就没事了，遇到过7到8次吧，最近跟进代码后暂时没遇到，先观望

coolsnowwolf, 2020-10-20:
供电跟不上的问题，看来还是别超频了

hxlls, 2020-10-20:
> 供电跟不上的问题，看来还是别超频了

测试了在1512下能稳定了，目前没发现问题，我上面放出的那个方式修改后测试可以正常了。PS:现在市面上好点的5V 3A micro usb 都不便宜，树莓派原装的70多，估计用这 玩意的人大部分都是5V 2A或者2.1A 左右的，所以应该1512MHZ 比较好

LissyTube, 2020-10-22:
5v 2.4

zgc, 2020-10-22:
绿联5v2.1a一直都是1.6，挺稳定的

liuyanantj, 2020-10-23:
刚刚测试了下 跑了coremark  第二次重启了  电源是小米的60w桌面电源  供电5V 最高应该是4.8A最高了

hxlls, 2020-10-23:
> 刚刚测试了下 跑了coremark 第二次重启了 电源是小米的60w桌面电源 供电5V 最高应该是4.8A最高了

是不是快充协议的，快充的话设备官方不建议用快充。。

ldada263, 2020-11-14:
5V2A平板电脑充电器一直稳定1.6

closed 2021-06-20 by coolsnowwolf