coolsnowwolf/lede #5681: R2S下风扇转个不停，这个好解决吗

hedahong, 2020-10-18:
发现风扇不停的在转（即使温度很低），之前用的friendlywrt的固件没有这个问题（cpu温度低的时候风扇会自动停，温度过热才会转），大佬们这个好解决吗？

chpos, 2020-10-23:
温度转个不停也不是什么坏事

hedahong, 2020-10-26:
能依据cpu温度 控制 启停 和 转速 就更完美了

qq1316237299, 2020-10-29:
bios下设置下

lifetyper, 2020-11-01:
> 
> 
> bios下设置下

你家r2s有bios？

hedahong, 2020-11-01:
:) 坐等大佬们来解决，友善的代码是否好移植过来

Tzwcard, 2021-01-05:
可以从friendlywrt的repo上拿脚本然后加到启动里面
https://github.com/friendlyarm/friendlywrt/blob/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh

hedahong, 2021-01-06:
> 可以从friendlywrt的repo上拿脚本然后加到启动里面
> https://github.com/friendlyarm/friendlywrt/blob/master-v19.07.1/target/linux/rockchip-rk3328/base-files/usr/bin/start-rk3328-pwm-fan.sh
> https://github.com/friendlyarm/friendlywrt/blob/master-v19.07.1/target/linux/rockchip-rk3328/base-files/etc/init.d/fa-rk3328-pwmfan
> 
> 把第一个.sh放到/usr/bin/内 加x权限 然后把第二个放到/etc/init.d/里面 也是加x权限
> 之后把它enable再start就行了
> 
> `/etc/init.d/fa-rk3328-pwmfan enable`
> `/etc/init.d/fa-rk3328-pwmfan start`

🐂B，谢谢。

closed 2021-06-20 by coolsnowwolf