coolsnowwolf/lede #5685: eth0无法nat

Zxilly, 2020-10-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：eth0无法NAT

### (2) 路由器型号和固件版本

A：HiWiFi HC5861 latest

### (3) 详细日志

A：
```
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.146508] ------------[ cut here ]------------
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.151295] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x26c/0x274
Tue Oct 20 09:40:18 2020 kern.info kernel: [ 2236.159756] NETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.167142] Modules linked in: rt2800soc rt2800mmio rt2800lib pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00soc rt2x00mmio rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY ts_kmp ts_fsm ts_bm slhc nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.167418]  asn1_decoder tcp_bbr ledtrig_usbport cryptodev xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic ecb authenc arc4 mmc_block mtk_sd mmc_core leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common aead cryptomgr crypto_null crypto_hash
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.315411] CPU: 0 PID: 0 Comm: swapper Not tainted 5.4.69 #0
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.321285] Stack : 805b0000 8055aef8 00000000 00000000 8055a0c8 87c0be14 8059015c 8058fd43
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.329833]         805071e4 00000000 806f33cc 805a0000 00200000 00000001 87c0bdc8 165075b3
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.338381]         00000000 00000000 80720000 00000000 746e6961 00000120 362e342e 30232039
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.346927]         00000000 0000006f 00000000 0004d013 00000000 00000009 00000000 8035a440
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.355473]         00000009 805a0000 00200000 00000122 00000000 8028d260 00000000 806f0000
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.364019]         ...
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.366526] Call Trace:
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.369045] [<8000b470>] show_stack+0x30/0x100
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.373615] [<800258e0>] __warn+0xc0/0x10c
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.377818] [<800259b8>] warn_slowpath_fmt+0x8c/0xac
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.382911] [<8035a440>] dev_watchdog+0x26c/0x274
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.387752] [<800675d8>] call_timer_fn.isra.34+0x20/0x90
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.393200] [<800677f0>] run_timer_softirq+0x1a8/0x1f4
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.398477] [<80483840>] __do_softirq+0x110/0x298
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.403305] [<80005f70>] except_vec_vi_end+0xb8/0xc4
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.408389] [<8048360c>] r4k_wait_irqoff+0x18/0x24
Tue Oct 20 09:40:18 2020 kern.warn kernel: [ 2236.413295] ---[ end trace 07d2ae260d82f8a2 ]---
Tue Oct 20 09:40:18 2020 kern.err kernel: [ 2236.418028] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Tue Oct 20 09:40:18 2020 kern.info kernel: [ 2236.424619] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Tue Oct 20 09:40:18 2020 kern.info kernel: [ 2236.431042] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06f0c000, max=1024, ctx=788, dtx=788, fdx=787, next=788
Tue Oct 20 09:40:18 2020 kern.info kernel: [ 2236.442332] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06508000, max=1024, calc=974, drx=979
Tue Oct 20 09:40:18 2020 kern.info kernel: [ 2236.459210] ralink_soc_eth 10100000.ethernet eth0: port 5 link down
Tue Oct 20 09:40:19 2020 daemon.notice netifd: Network device 'eth0' link is down
Tue Oct 20 09:40:19 2020 kern.info kernel: [ 2236.479640] br-lan: port 1(eth0.1) entered disabled state
Tue Oct 20 09:40:19 2020 daemon.notice netifd: VLAN 'eth0.1' link is down
Tue Oct 20 09:40:19 2020 daemon.notice netifd: VLAN 'eth0.2' link is down
Tue Oct 20 09:40:19 2020 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Oct 20 09:40:19 2020 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Tue Oct 20 09:40:19 2020 daemon.notice netifd: Interface 'wan6' is now down
Tue Oct 20 09:40:19 2020 daemon.notice netifd: wan (1627): udhcpc: received SIGTERM
Tue Oct 20 09:40:19 2020 daemon.notice netifd: wan (1627): udhcpc: unicasting a release of 10.73.23.56 to 10.73.0.1
Tue Oct 20 09:40:19 2020 daemon.notice netifd: wan (1627): udhcpc: sending release
Tue Oct 20 09:40:19 2020 daemon.notice netifd: wan (1627): udhcpc: entering released state
Tue Oct 20 09:40:19 2020 daemon.notice netifd: wan (1627): Command failed: Permission denied
Tue Oct 20 09:40:19 2020 daemon.notice netifd: Interface 'wan' is now down
Tue Oct 20 09:40:19 2020 daemon.err miniupnpd[2464]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Oct 20 09:40:19 2020 daemon.err miniupnpd[2464]: Failed to get IP for interface eth0.2
Tue Oct 20 09:40:19 2020 daemon.warn miniupnpd[2464]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Tue Oct 20 09:40:20 2020 kern.info kernel: [ 2237.778411] br-lan: port 1(eth0.1) entered blocking state
Tue Oct 20 09:40:20 2020 kern.info kernel: [ 2237.784002] br-lan: port 1(eth0.1) entered forwarding state
Tue Oct 20 09:40:20 2020 daemon.notice netifd: Network device 'eth0' link is up
Tue Oct 20 09:40:20 2020 daemon.notice netifd: VLAN 'eth0.1' link is up
Tue Oct 20 09:40:20 2020 daemon.notice netifd: VLAN 'eth0.2' link is up
Tue Oct 20 09:40:20 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Oct 20 09:40:20 2020 daemon.notice netifd: Interface 'wan' is setting up now
Tue Oct 20 09:40:20 2020 daemon.notice netifd: Interface 'wan6' has link connectivity
Tue Oct 20 09:40:20 2020 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Oct 20 09:40:20 2020 daemon.notice netifd: Interface 'wan6' is now up
Tue Oct 20 09:40:21 2020 kern.info kernel: [ 2238.159229] ralink_soc_eth 10100000.ethernet eth0: port 5 link up (100Mbps/Full duplex)
Tue Oct 20 09:40:21 2020 daemon.notice netifd: wan (24840): udhcpc: started, v1.31.1
Tue Oct 20 09:40:21 2020 daemon.notice netifd: wan (24840): udhcpc: sending discover
Tue Oct 20 09:40:23 2020 daemon.notice netifd: wan (24840): udhcpc: sending discover
Tue Oct 20 09:40:26 2020 daemon.notice netifd: wan (24840): udhcpc: sending discover
Tue Oct 20 09:40:26 2020 daemon.notice netifd: wan (24840): udhcpc: sending select for 10.73.23.56
Tue Oct 20 09:40:26 2020 daemon.notice netifd: wan (24840): udhcpc: lease of 10.73.23.56 obtained, lease time 259200
Tue Oct 20 09:40:26 2020 daemon.notice netifd: Interface 'wan' is now up
Tue Oct 20 09:40:31 2020 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)
Tue Oct 20 09:40:32 2020 daemon.info privoxy[7569]: Restart request due to 'ifup' of interface 'wan6'
Tue Oct 20 09:40:33 2020 daemon.notice privoxy[7569]: Shutdown successfully
Tue Oct 20 09:40:35 2020 daemon.notice privoxy[25243]: Started successfully
Tue Oct 20 09:40:46 2020 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
Tue Oct 20 09:40:48 2020 daemon.info privoxy[25243]: Restart request due to 'ifup' of interface 'wan'
Tue Oct 20 09:40:49 2020 daemon.notice privoxy[25243]: Shutdown successfully
Tue Oct 20 09:40:51 2020 daemon.notice privoxy[25568]: Started successfully
```

closed 2020-10-20 by coolsnowwolf

Zxilly, 2020-10-20:
@coolsnowwolf 额，能提供一点建议吗？
这个`transmit queue 0 timed out`的问题查了很多文档都没有解决，而且我的两台设备都有这个问题

AmadeusGhost, 2020-10-21:
You can reopened this issue.