coolsnowwolf/lede #5688: 有没有用R2S的朋友，R2S外接无线网卡能做无线中继吗

xiakeng, 2020-10-20:
经常出差，想要备一个小一点的旅行滤油器，能爬梯子的那种。

无线中继的情况下
家里有HC5962，有点大，翻梯子下行30M上行10M的水平
买过AR750，体积很小正合我意，但是性能太拉了，翻梯子大概下行15M上行3M

R2S大小完美，性能更好，想入，但是不确定再配一个无线网卡插着后能不能做无线中继。毕竟现在酒店或民宿不一定能弄跟网线做AP。
网上找了一圈也就找到有线中继AP的形式。

有没有朋友分享下经验

programmejava, 2020-10-21:
我用的rtl8812au做无线，限于usb2.0的原因，wifi是usb3.0也只能用2.0，而且只能做单频，2.4G和5G只能选一，某些信道不能启动wifi， 如果wifi做client中继，wifi就只能做client没有AP功能了.   但可以用lan  。所以这个R2S这个做中继只能做一半, 后一半只能通过有线完成了。

VergilGao, 2020-10-22:
你这个方案没用几次就要丢，很容易忘在酒店的。

tjaol, 2020-10-22:
我也是想要这种方案，WiFi中继场景很多，反而插线的场景少，最多就是WAN口插线，LAN口插线很麻烦

xiakeng, 2020-10-22:
> 我用的rtl8812au做无线，限于usb2.0的原因，wifi是usb3.0也只能用2.0，而且只能做单频，2.4G和5G只能选一，某些信道不能启动wifi， 如果wifi做client中继，wifi就只能做client没有AP功能了. 但可以用lan 。所以这个R2S这个做中继只能做一半, 后一半只能通过有线完成了。

谢谢，不入了不入了

closed 2020-10-22 by xiakeng

orut34iop, 2021-01-05:
> 经常出差，想要备一个小一点的旅行滤油器，能爬梯子的那种。
> 
> 无线中继的情况下
> 家里有HC5962，有点大，翻梯子下行30M上行10M的水平
> 买过AR750，体积很小正合我意，但是性能太拉了，翻梯子大概下行15M上行3M
> 
> R2S大小完美，性能更好，想入，但是不确定再配一个无线网卡插着后能不能做无线中继。毕竟现在酒店或民宿不一定能弄跟网线做AP。
> 网上找了一圈也就找到有线中继AP的形式。
> 
> 有没有朋友分享下经验

我刚刚试过, R2S配合USB无线网卡做无线中继,  用途是外面旅游的时候, 把酒店的无线网络转换为翻墙后的新的无线网络(即无线中继+ v2ray翻墙), 验证下来功能虽然可用但不稳定不推荐使用, 主要原因不细说了,归结是R2S固件对这个功能支持还不完善; 后来换成有线转NAT热点模式(WAN口接酒店有线网口+ AP热点 + V2ray翻墙),  终于可以在酒店看youtube和Netflix了.