coolsnowwolf/lede #5689: 竞斗云2.0固件的交换机创建存在冲突

wattic, 2020-10-20:
1.竞斗云2.0固件中，交换机设置不包括wan口，如果配置新的vlan，则新vlan默认为vlan2，会和wan口实际使用的vlan2(?)产生冲突，导致wan口和新vlan都不可用,在/etc/config/network中手工更改新vlan为vlan3可以解决该问题。
2.此外，由于未知原因，创建vlan3以后（原来在eth0上的br-lan自动转到eth0.1），通过网线连接在以太网交换机eth0.1上的客户端无法上线（设置静态ip也不能），连接在wlan上的客户端则可以正常连接路由器。该问题无法通过重新设置eth0.3的mac地址解决。

经过搜索，在另一个固件中存在类似的问题： https://github.com/Lienol/openwrt/issues/274； 未知是否解决（可能由问题1导致的）问题2。
问题1的描述可以直接搬运这位的图：
[image]

固件版本：OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222) 内核4.19.138
感谢！

coolsnowwolf, 2020-10-21:
你认为有问题的那个是 官方 Op 的， WAN 模拟为单独的 eth

你认为正常的是我打的补丁，大家喜欢官方的，所以统一到官方了

wattic, 2020-10-21:
> 你认为有问题的那个是 官方 Op 的， WAN 模拟为单独的 eth
感谢大佬回复，搜到了openwrt官方讨论的，好像4.19(?)开始IPQ40XX的这部分功能就实质上坏了
从他们扯皮的程度来看是要等个厂商驱动，不知道要猴年马月了。

> 你认为正常的是我打的补丁，大家喜欢官方的，所以统一到官方了
好像论坛上是有人给了dual tagging的修复补丁但是对于只会github自动编译和折腾bash的人，一下子进化到打patch似乎有点难。
话说喜欢官方的人可以自己去拉原汁原味官方源嘛，再加俩服务一点都不难连我都会。
问题是官方源是做不到对绝大多数国内热门固件的维护的

wattic, 2020-10-21:
额好像改mac和vlan tag的要给内核打补丁。。。
然后我连内核在哪都不知道。。

llfj, 2021-02-14:
我也支持，老大重新打补丁。

closed 2021-06-20 by coolsnowwolf