coolsnowwolf/lede #5693: Compilation Errors led by flow offload code of mediatek.

Chi-Tan-Da-Eru, 2020-10-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
No more details.
### (1) 具体问题

A：Multiple Definition in codes of mtwifi, which makes the compilation errors happen.

### (2) 路由器型号和固件版本

A：Xiaomi Redmi Router AC2100

### (3) 详细日志

A：```
cp -fpR /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips
chmod: cannot access '/sbin/mtkwifi': No such file or directory
/home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/nginx: line 10: /usr/bin/nginx-util: No such file or directory
Enabling autoreboot
Enabling boot
Enabling bootcount
Enabling cron
Enabling ddns
Enabling dnsmasq
Enabling done
Enabling dropbear
Enabling firewall
chmod: cannot access '/sbin/mtkwifi': No such file or directory
Enabling firmware.sh
Enabling fstab
Enabling gpio_switch
Enabling led
Enabling log
Enabling network
./etc/init.d/nginx: line 10: /usr/bin/nginx-util: No such file or directory
Enabling nginx
Enabling nlbwmon
Enabling rpcd
Enabling set-irq-affinity
Enabling sfe
Enabling shadowsocksr
Enabling shortcut-fe
Enabling sysctl
Enabling sysfixtime
Enabling sysntpd
Enabling system
Enabling uhttpd
Enabling umount
Enabling unblockmusic
Enabling urandom_seed
Enabling urngd
Enabling uwsgi
Enabling zram
sed -i "s/Installed-Time: .*/Installed-Time: 1603263141/" /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/status
rm -rf /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/tmp/*
rm -f /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/lists/*
rm -f /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/*.postinst*
rm -f /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/var/lock/*.lock
rm -rf /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/boot
find /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/root-ramips/ -mindepth 1 -execdir touch -hcd "@1603263141" "{}" +
make[2]: Leaving directory '/home/chitanda/lede'
make[2]: Entering directory '/home/chitanda/lede'
make[3]: Entering directory '/home/chitanda/lede/target/linux'
make[4]: Entering directory '/home/chitanda/lede/target/linux/ramips'
rm -f /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/localversion
[... 17 lines cut ...]
make -C /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72 KCFLAGS="-ffile-prefix-map=/home/chitanda/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/chitanda/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Oct 21 06:52:21 2020" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/chitanda/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/chitanda/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=5.4.72 CC="mipsel-openwrt-linux-musl-gcc" all modules
make[5]: Entering directory '/home/chitanda/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
mipsel-openwrt-linux-musl-ld: drivers/net/ethernet/ralink/mtk_offload.o: in function `mtk_offload_check_rx':
target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/ralink/mtk_offload.c:508: multiple definition of `mtk_offload_check_rx'; drivers/net/ethernet/mediatek/mtk_offload.o:target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/mediatek/mtk_offload.c:122: first defined here
mipsel-openwrt-linux-musl-ld: drivers/net/ethernet/ralink/mtk_debugfs.o: in function `mtk_ppe_debugfs_init':
target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/ralink/mtk_debugfs.c:103: multiple definition of `mtk_ppe_debugfs_init'; drivers/net/ethernet/mediatek/mtk_ppe_debugfs.o:target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/mediatek/mtk_ppe_debugfs.c:104: first defined here
make[5]: *** [Makefile:1103: vmlinux] Error 1
make[5]: Leaving directory '/home/chitanda/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72'
make[4]: *** [Makefile:26: /home/chitanda/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/.image] Error 2
make[4]: Leaving directory '/home/chitanda/lede/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/chitanda/lede/target/linux'
time: target/linux/install#6.42#4.14#10.42
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/home/chitanda/lede'
make[1]: *** [target/Makefile:19: /home/chitanda/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/chitanda/lede'
make: *** [/home/chitanda/lede/include/toplevel.mk:224：world] 错误 2
```

Cathgao, 2020-10-21:
最新commit同样的错误
```
2020-10-21T10:05:40.7795031Z mipsel-openwrt-linux-musl-ld: drivers/net/ethernet/ralink/mtk_offload.o: in function `mtk_offload_check_rx':
2020-10-21T10:05:40.7804506Z target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/ralink/mtk_offload.c:508: multiple definition of `mtk_offload_check_rx'; drivers/net/ethernet/mediatek/mtk_offload.o:target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/mediatek/mtk_offload.c:122: first defined here
2020-10-21T10:05:40.7813540Z mipsel-openwrt-linux-musl-ld: drivers/net/ethernet/ralink/mtk_debugfs.o: in function `mtk_ppe_debugfs_init':
2020-10-21T10:05:40.7822521Z target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/ralink/mtk_debugfs.c:103: multiple definition of `mtk_ppe_debugfs_init'; drivers/net/ethernet/mediatek/mtk_ppe_debugfs.o:target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/mediatek/mtk_ppe_debugfs.c:104: first defined here
2020-10-21T10:05:42.0269201Z Makefile:1103: recipe for target 'vmlinux' failed
2020-10-21T10:05:42.0272716Z make[5]: *** [vmlinux] Error 1
2020-10-21T10:05:42.0273852Z make[5]: Leaving directory '/home/runner/work/newifi3/newifi3/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72'
2020-10-21T10:05:42.0275707Z Makefile:25: recipe for target '/home/runner/work/newifi3/newifi3/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/.image' failed
2020-10-21T10:05:42.0277036Z make[4]: Leaving directory '/home/runner/work/newifi3/newifi3/lede/target/linux/ramips'
2020-10-21T10:05:42.0278000Z Makefile:13: recipe for target 'install' failed
2020-10-21T10:05:42.0278949Z make[3]: Leaving directory '/home/runner/work/newifi3/newifi3/lede/target/linux'
2020-10-21T10:05:42.0279667Z time: target/linux/install#646.33#65.34#363.03
2020-10-21T10:05:42.0297907Z make[4]: *** [/home/runner/work/newifi3/newifi3/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/.image] Error 2
2020-10-21T10:05:42.0298868Z target/Makefile:23: recipe for target 'target/linux/install' failed
2020-10-21T10:05:42.0299704Z make[3]: *** [install] Error 2
2020-10-21T10:05:42.0300472Z make[2]: *** [target/linux/install] Error 2
2020-10-21T10:05:42.0302369Z make[2]: Leaving directory '/home/runner/work/newifi3/newifi3/lede'
2020-10-21T10:05:42.0319885Z target/Makefile:19: recipe for target '/home/runner/work/newifi3/newifi3/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
2020-10-21T10:05:42.0341340Z make[1]: *** [/home/runner/work/newifi3/newifi3/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
2020-10-21T10:05:42.0352644Z make[1]: Leaving directory '/home/runner/work/newifi3/newifi3/lede'
2020-10-21T10:05:42.0367910Z Build failed - please re-run with -j1 to see the real error message
2020-10-21T10:05:42.0369940Z /home/runner/work/newifi3/newifi3/lede/include/toplevel.mk:222: recipe for target 'world' failed
2020-10-21T10:05:42.0375229Z make: *** [world] Error 1
2020-10-21T10:05:42.0467809Z ##[error]Process completed with exit code 2.
```

blueveryday, 2020-10-21:
似的，我也有类似的错误。

> 
  AR      arch/mips/math-emu/lib.a
  EXPORTS arch/mips/math-emu/lib-ksyms.o
  AR      arch/mips/math-emu/built-in.a
  AR      virt/lib/built-in.a
  AR      virt/built-in.a
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
mipsel-openwrt-linux-musl-ld: drivers/net/ethernet/ralink/mtk_offload.o: in function `mtk_offload_check_rx':
target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/ralink/mtk_offload.c:508: multiple definition of `mtk_offload_check_rx'; drivers/net/ethernet/mediatek/mtk_offload.o:target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/mediatek/mtk_offload.c:122: first defined here
mipsel-openwrt-linux-musl-ld: drivers/net/ethernet/ralink/mtk_debugfs.o: in function `mtk_ppe_debugfs_init':
target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/ralink/mtk_debugfs.c:103: multiple definition of `mtk_ppe_debugfs_init'; drivers/net/ethernet/mediatek/mtk_ppe_debugfs.o:target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/drivers/net/ethernet/mediatek/mtk_ppe_debugfs.c:104: first defined here
make[5]: *** [Makefile:1103: vmlinux] Error 1
make[5]: Leaving directory '/home/ago/Desktop/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72'
make[4]: *** [Makefile:26: /home/ago/Desktop/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.72/.image] Error 2
make[4]: Leaving directory '/home/ago/Desktop/lede/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/ago/Desktop/lede/target/linux'
time: target/linux/install#659.29#62.61#714.83
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/home/ago/Desktop/lede'
make[1]: *** [target/Makefile:19: /home/ago/Desktop/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ago/Desktop/lede'
make: *** [/home/ago/Desktop/lede/include/toplevel.mk:224：world] 错误 2

>

jxph1123, 2020-10-21:
重新make clean试试

Chi-Tan-Da-Eru, 2020-10-21:
> 重新make clean试试

全新克隆的，并不是二次编译。

HoldOnBro, 2020-10-21:
同

AmadeusGhost, 2020-10-21:
草，2个不同的 hwnat 实现代码冲突了 ,,,

blueveryday, 2020-10-21:
> 草，2个不同的 hwnat 实现代码冲突了 ,,,

加油！

PR #5697 (2020-10-23): ramips: ralink: rename some functions
fix #5693, fix #5695 , fix #5696 , fix #5702
closed 2020-10-23 by coolsnowwolf