coolsnowwolf/lede #5694: 编译出错，麻烦看下什么原因 package/feeds/helloworld/naiveproxy/compile Error

keeno1982, 2020-10-21:
make[3]: Entering directory '/home/inter/lede/feeds/helloworld/naiveproxy'
rm -f /home/inter/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-86.0.4240.75-1/.built
touch /home/inter/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-86.0.4240.75-1/.built_check
( cd /home/inter/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-86.0.4240.75-1 ; ./tools/import-upstream.sh ; . ./init_env.sh "aarch64" "sunxi" "cortex-a53" "" "/home/inter/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl" "/home/inter/lede/dl"; export naive_flags="${naive_flags} ${naive_ccache_flags}" ; export OP_STAGING_DIR="/home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl" ; mkdir -p out ; ./gn/out/gn gen "out/Release" --args="${naive_flags}" --script-executable="/home/inter/lede/staging_dir/hostpkg/bin/python3" ; /home/inter/lede/staging_dir/hostpkg/bin/ninja -C "out/Release" naive ; )
+ + tr \n .
+ cut -d. -f1-4
cut -d= -f2 src/chrome/VERSION
+ have_version=86.0.4240.75
+ cat CHROMIUM_VERSION
+ want_version=86.0.4240.75
+ [ 86.0.4240.75 = 86.0.4240.75 ]
+ exit 0
Done. Made 16631 targets from 2416 files in 125002ms
ninja: Entering directory `out/Release'
[1/2061] CXX obj/buildtools/third_party/libc++/libc++/charconv.o
FAILED: obj/buildtools/third_party/libc++/libc++/charconv.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/charconv.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=__attribute__\(\(__visibility__\(\"default\"\)\)\) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-12-init-3492-ga1caa302-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=dc0edfa18ab4467282be4ca4b3b7df7d1a4ff81c -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/ignores.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -fstrict-aliasing -fPIC -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-psabi -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-string-concatenation -Wno-max-tokens -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl -I/home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/nss -I/home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/nspr -fvisibility-inlines-hidden --target=aarch64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../buildtools/third_party/libc++/trunk/src/charconv.cpp -o obj/buildtools/third_party/libc++/libc++/charconv.o
error: ../../chrome/android/profiles/afdo.prof: Could not open profile: Unrecognized sample profile encoding format
1 error generated.
[2/2061] CXX obj/buildtools/third_party/libc++/libc++/any.o
FAILED: obj/buildtools/third_party/libc++/libc++/any.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/any.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=__attribute__\(\(__visibility__\(\"default\"\)\)\) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-12-init-3492-ga1caa302-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=dc0edfa18ab4467282be4ca4b3b7df7d1a4ff81c -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -fcomplete-member-pointers -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fsanitize=cfi-vcall -fsanitize-blacklist=../../tools/cfi/ignores.txt -fsanitize=cfi-icall -fvisibility=hidden -Xclang -add-plugin -Xclang find-bad-constructs -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -fstrict-aliasing -fPIC -Wall -Wno-unused-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-psabi -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-string-concatenation -Wno-max-tokens -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -nostdinc++ -isystem../../buildtools/third_party/libc++/trunk/include -isystem../../buildtools/third_party/libc++abi/trunk/include --sysroot=../../../../../../staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl -I/home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/nss -I/home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl/usr/include/nspr -fvisibility-inlines-hidden --target=aarch64-openwrt-linux-musl -D_LIBCPP_HAS_MUSL_LIBC -D__UCLIBC__ -c ../../buildtools/third_party/libc++/trunk/src/any.cpp -o obj/buildtools/third_party/libc++/libc++/any.o
error: ../../chrome/android/profiles/afdo.prof: Could not open profile: Unrecognized sample profile encoding format
1 error generated.
[3/2061] CXX obj/buildtools/third_party/libc++/libc++/bind.o
FAILED: obj/buildtools/third_party/libc++/libc++/bind.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/buildtools/third_party/libc++/libc++/bind.o.d -D_LIBCPP_BUILDING_LIBRARY -D_LIBCPP_OVERRIDABLE_FUNC_VIS=__attribute__\(\(__visibility__\(\"default\"\)\)\) -DLIBCXX_BUILDING_LIBCXXABI -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DCR_CLANG_REVISION=\"llvmorg-12-init-3492-ga1caa302-1\" -D_LIBCPP_ABI_UNSTABLE -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCXXABI_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_NODISCARD -DCR_LIBCXX_REVISION=375504 -DCR_SYSROOT_HASH=dc0edfa18ab4467282be4ca4b3b7df7d1a4ff81c -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fprofile-sample-use=../../chrome/android/profiles/afdo.prof -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-ta [... cut ...]
make[3]: Leaving directory '/home/inter/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#17.54#4.18#137.64
make[2]: *** [package/Makefile:113: package/feeds/helloworld/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/inter/lede'
make[1]: *** [package/Makefile:107: /home/inter/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/inter/lede'
make: *** [/home/inter/lede/include/toplevel.mk:224: world] Error 2

blueveryday, 2020-10-21:
一样的，新代码过不了。

keeno1982, 2020-10-21:
搞定，把配置文件里面的naiveproxy相关的注释掉就可以了

greennyreborn, 2020-10-21:
据说是naiveproxy编译需要很大的内存。
不用的话 menuconfig里面去掉naiveproxy就可以了

realsteam, 2020-10-22:
需要用到naiveproxy，怎么办？一直编译不过

afeiy, 2020-10-22:
> 需要用到naiveproxy，怎么办？一直编译不过
naiveproxy是个好协议，不过短期内恐怕注定小众，编译环境需要大内存，大硬盘，开发者又无ipk供下载。
用过一段时间，转木马了。

realsteam, 2020-10-22:
> > 需要用到naiveproxy，怎么办？一直编译不过
> > naiveproxy是个好协议，不过短期内恐怕注定小众，编译环境需要大内存，大硬盘，开发者又无ipk供下载。
> > 用过一段时间，转木马了。

请问下到底需要多大内存和硬盘？搞这么厉害？

afeiy, 2020-10-22:
> > > 需要用到naiveproxy，怎么办？一直编译不过
> > > naiveproxy是个好协议，不过短期内恐怕注定小众，编译环境需要大内存，大硬盘，开发者又无ipk供下载。
> > > 用过一段时间，转木马了。
> 
> 请问下到底需要多大内存和硬盘？搞这么厉害？
我以前16G内存，100G硬盘可以过，但是这次更新后依然不过。

keeno1982, 2020-10-22:
> 
> 
> > > > 需要用到naiveproxy，怎么办？一直编译不过
> > > > naiveproxy是个好协议，不过短期内恐怕注定小众，编译环境需要大内存，大硬盘，开发者又无ipk供下载。
> > > > 用过一段时间，转木马了。
> > 
> > 
> > 请问下到底需要多大内存和硬盘？搞这么厉害？
> > 我以前16G内存，100G硬盘可以过，但是这次更新后依然不过。

太夸张了，惹不起

realsteam, 2020-10-22:
> > > > 需要用到naiveproxy，怎么办？一直编译不过
> > > > naiveproxy是个好协议，不过短期内恐怕注定小众，编译环境需要大内存，大硬盘，开发者又无ipk供下载。
> > > > 用过一段时间，转木马了。
> > 
> > 
> > 请问下到底需要多大内存和硬盘？搞这么厉害？
> > 我以前16G内存，100G硬盘可以过，但是这次更新后依然不过。

作者故意为之？还是不得不如此？那要是这需要变态的硬件配置干脆去掉得了

Cathgao, 2020-10-22:
这配置合着是编译一个完整chromium吧？

Dimitri2020007, 2020-10-22:
naiveproxy的软件版本更新了补丁没更新，新naiveproxy要装一个新的软件才能过。也是运行库。

outemancut, 2020-10-23:
> naiveproxy的软件版本更新了补丁没更新，新naiveproxy要装一个新的软件才能过。也是运行库。

怎么搞 求解决！

realsteam, 2020-10-23:
> naiveproxy的软件版本更新了补丁没更新，新naiveproxy要装一个新的软件才能过。也是运行库。

FAILED: obj/buildtools/third_party/libc++/libc++/algorithm.o 
是这个吗？如何解决？

1715173329, 2020-10-23:
你自己梯子问题罢了，没什么别的原因

lxwq, 2020-10-23:
sudo apt install libtinfo5试试,
我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,

iKenzo, 2020-10-23:
> sudo apt install libtinfo5试试,
> 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,

编译成功，非常感谢！

realsteam, 2020-10-23:
> sudo apt install libtinfo5试试,
> 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,

搞定了，谢谢指点。

afeiy, 2020-10-23:
> sudo apt install libtinfo5试试,
> 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,

确实如此

Dimitri2020007, 2020-10-23:
> > naiveproxy的软件版本更新了补丁没更新，新naiveproxy要装一个新的软件才能过。也是运行库。
> 
> 怎么搞 求解决！

ubuntu安装这个libtinfo5

blueveryday, 2020-10-23:
需要编译的软件包没有下载完整就会出现这个情况。
terminal走的网络不会从电脑的全局代理服务器走下载，（假设你已经开了Qv2ray这类软件）

解决方法：
启动电脑的代理服务器，比如Qv2ray这类软件。
安装如下工具：
sudo apt-get install proxychains

进入/etc下面，修改proxychains的配置文件config。
有两个文件，一个是proxychains，一个是proxychains4的配置文件，我都改成本地电脑Qv2ray的socks5的127.0.0.1:1089这个地址。

然后在terminal界面，执行命令：
proxychains4 make -j1 V=s
就行了。
这时候terminal就强制走127.0.0.1:1089这个socks5本地代理地址了。
等走完编译流程，这时候在编译naiveproxy的时候，所需要下载的软件包就能够走代理下载了。
只要下载文件不失败，不出错，就可以顺利编译好固件了。

亲测有效。

OleStore, 2020-10-23:
> sudo apt install libtinfo5试试,
> 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,

确定是依赖问题

keeno1982, 2020-10-23:
> 
> 
> sudo apt install libtinfo5试试,
> 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,

装了 libtinfo5 还是报错，FAILED: obj/buildtools/third_party/libc++/libc++/charconv.o，这个是要装那个依赖？

blueveryday, 2020-10-23:
> > sudo apt install libtinfo5试试,
> > 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,
> 
> 装了 libtinfo5 还是报错，FAILED: obj/buildtools/third_party/libc++/libc++/charconv.o，这个是要装那个依赖？

缺软件包，删除对应目录，重新下载软件包。

keeno1982, 2020-10-23:
> 
> 
> > > sudo apt install libtinfo5试试,
> > > 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,
> > 
> > 
> > 装了 libtinfo5 还是报错，FAILED: obj/buildtools/third_party/libc++/libc++/charconv.o，这个是要装那个依赖？
> 
> 缺软件包，删除对应目录，重新下载软件包。

要删哪个目录呢，没找到对应的路径

keeno1982, 2020-10-23:
> 
> 
> > > sudo apt install libtinfo5试试,
> > > 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,
> > 
> > 
> > 装了 libtinfo5 还是报错，FAILED: obj/buildtools/third_party/libc++/libc++/charconv.o，这个是要装那个依赖？
> 
> 缺软件包，删除对应目录，重新下载软件包。

赞，通过了！

keeno1982, 2020-10-23:
> 
> 
> > naiveproxy的软件版本更新了补丁没更新，新naiveproxy要装一个新的软件才能过。也是运行库。
> 
> FAILED: obj/buildtools/third_party/libc++/libc++/algorithm.o
> 是这个吗？如何解决？

删掉 /lede/build_dir/target-aarch64_cortex-a53_musl 目录下所有开头naiveproxy的文件夹
再删掉 /lede/dl 目录下所有开头是naiveproxy的文件
重新编译就OK了

outemancut, 2020-10-24:
我是把虚拟机搞到8G内存才编译过了

closed 2020-11-02 by coolsnowwolf

wanyanjiabin, 2020-11-13:
> sudo apt install libtinfo5试试,
> 我也是搞了几天,今天在这翻lssues,看到有人说要装这个依赖包,然后就编译正常了,

必须要赞你一下,ubuntu系统原始编译openwrt正常,引入三方s插件之后就需要这个了