coolsnowwolf/lede #5699: 能否加入airconnect模块

ragelancer, 2020-10-22:
你好，L大。
airconnect是一个将音频通过airplay串流到不支持airplay的dlna/upnp设备上的软件。
详细介绍见 https://github.com/philippe44/AirConnect
请问能否将其功能整合进openwrt里呢。

coolsnowwolf, 2020-10-23:
[image]

ragelancer, 2020-10-23:
多谢L大，那请问在编译的时候选择哪个组件，没有找到类似luci-app-airconnect这样名称的组件。

sxx1314, 2020-10-24:
这个项目多贴心，直接预编译好了。  直接下载二进制写好config运行即可。另外docker也有现成镜像。

ragelancer, 2020-10-25:
> 这个项目多贴心，直接预编译好了。 直接下载二进制写好config运行即可。另外docker也有现成镜像。

我把预编译好的文件放在了/bin目录下，也设置了运行权限
但运行后提示
`-ash: ./airupnp-aarch64: not found
`请问这是怎么回事？

closed 2021-06-20 by coolsnowwolf

erdrd, 2021-06-30:
@coolsnowwolf 
Since you seem to have had some success with AirConnect on OpenWrt and a similar setup, I assume, I have two questions:

1. On my router (Espressobin, Marvell Armada 88F3700 based), I can run the airupnp-aarch64-static binary fine, but do not discover any devices. looks like multicast interference on the router itself (since the the rest of the network has no issues; I can run a docker container on unraid with airconnect just fine, but would prefer to have it run on the router). 
Any settings/configs that are non-obvious?

2. where can I find your luci-app-airconnect? :)

Thanks
A