coolsnowwolf/lede #5700: 请问没有生成/tmp/localtime这个文件是哪里的原因

gaoyaxuan, 2020-10-22:
root@OpenWrt:~# ll /etc/localtime
lrwxrwxrwx    1 root     root            14 Oct 21 14:52 /etc/localtime -> /tmp/localtime
root@OpenWrt:~# cat /tmp/localtime
cat: can't open '/tmp/localtime': No such file or directory
因为docker需要挂载这个文件来更改容器内部的时间.求赐教

VergilGao, 2020-10-23:
如果你的docker镜像没有配置时区的环境变量，那就把这俩文件以只读方式挂载进去。
/etc/localtime
/etc/timezone

gaoyaxuan, 2020-10-24:
> 如果你的docker镜像没有配置时区的环境变量，那就把这俩文件以只读方式挂载进去。
> /etc/localtime
> /etc/timezone

问题是,这两个文件根本不存在!/etc/localtime指向了/tmp/localtime,而/tmp/localtime文件根本不存在的.

VergilGao, 2020-10-24:
> > 如果你的docker镜像没有配置时区的环境变量，那就把这俩文件以只读方式挂载进去。
> > /etc/localtime
> > /etc/timezone
> 
> 问题是,这两个文件根本不存在!/etc/localtime指向了/tmp/localtime,而/tmp/localtime文件根本不存在的.

奥，查了一下，openwrt用的是uclibc，时区文件是 /etc/TZ
这个就没办法了，除非你切换成glibc，然后手动配置。 <- 大坑

最简单的办法就是自己打包一层，手动安装timezone 然后 ln -sf /usr/share/zoneinfo/Asia/Shanghai /etc/localtime

gaoyaxuan, 2020-10-24:
> 奥，查了一下，openwrt用的是uclibc，时区文件是 /etc/TZ
> 这个就没办法了，除非你切换成glibc，然后手动配置。 <- 大坑
>
> 最简单的办法就是自己打包一层，手动安装timezone 然后 ln -sf /usr/share/zoneinfo/Asia/Shanghai /etc/localtime

> 奥，查了一下，openwrt用的是uclibc，时区文件是 /etc/TZ
> 这个就没办法了，除非你切换成glibc，然后手动配置。 <- 大坑
>
> 最简单的办法就是自己打包一层，手动安装timezone 然后 ln -sf /usr/share/zoneinfo/Asia/Shanghai /etc/localtime

好的!谢谢大佬!

acooler15, 2021-02-07:
安装`zoneinfo-asia`包，然后运行命令`/etc/init.d/system reload`

[image]

closed 2021-06-20 by coolsnowwolf

MrGengCom, 2025-03-09:
> 安装`zoneinfo-asia`包，然后运行命令`/etc/init.d/system reload`
> 
> [image]

Great, solve my problem.