coolsnowwolf/lede #5707: 编译星际宝盒一直出问题

itpleb, 2020-10-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：以前编译过星际宝盒，现在重新用最新版本就编译不过。一直几天都是这样。

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：de/inspector/Runtime.h'
tools/v8_gypfiles/v8_base_without_compiler.target.mk:26: warning: overriding recipe for target '/home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
tools/v8_gypfiles/v8_base_without_compiler.host.mk:26: warning: ignoring old recipe for target '/home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
  LD_LIBRARY_PATH=/home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/lib.host:/home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated; python ../../deps/v8/tools/run.py "/home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/bytecode_builtins_list_generator" "/home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h"
Traceback (most recent call last):
  File "../../deps/v8/tools/run.py", line 12, in <module>
    sys.exit(subprocess.call(sys.argv[1:]))
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
make[5]: *** [tools/v8_gypfiles/generate_bytecode_builtins_list.target.mk:13: /home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h] Error 1
make[4]: *** [Makefile:101: node] Error 2
make[4]: Leaving directory '/home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1'
make[3]: *** [Makefile:157: /home/itpleb/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.16.1/.built] Error 2
make[3]: Leaving directory '/home/itpleb/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#0.23#1.42#1.87
make[2]: *** [package/Makefile:113: package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/itpleb/lede'
make[1]: *** [package/Makefile:107: /home/itpleb/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/itpleb/lede'
make: *** [/home/itpleb/lede/include/toplevel.mk:224: world] Error 2

AmadeusGhost, 2020-10-25:
解决了那不关闭 issue？

itpleb, 2020-10-25:
> 解决了那不关闭 issue？
重新编译发现又是出这个target-arm_cortex-a7+neon-vfpv4_musl_eab,2019时用WSL1编译都没有问题，就是下面的版本，云音乐功能也有。现在用WSL2和github编译也是一样有问题

主机名 | OpenWrt
-- | --
主机型号 | MobiPromo CM520-79F (CpuMark : 6804.456919 Scores)
架构 | ARMv7 Processor rev 5 (v7l) x 4
固件版本 | OpenWrt R20.3.19 / LuCI Master (git-20.087.30210-56fe7ae)
内核版本 | 4.14.172
本地时间 | Sun Oct 25 13:27:16 2020
运行时间 | 18h 49m 13s
平均负载 | 0.21, 0.26, 0.26
CPU 使用率（%） | 10 %

itpleb, 2020-10-25:
用github actions解决了。

closed 2020-10-25 by itpleb