coolsnowwolf/lede #5712: R2S断流

ostrichii, 2020-10-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
R2S在300M速度PT下载时出现断流
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
R2S测试了好几次，挂下PT下载过一段时间会出现断流，然后恢复，测试几次都是同样的问题
### (1) 具体问题
R2S测试了好几次，挂下PT下载过一段时间会出现断流，然后恢复，测试几次都是同样的问题
[image]
[image]

### (2) 路由器型号和固件版本
A：
R2S
### (3) 详细日志

A：Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.969183] ------------[ cut here ]------------
Sun Oct 25 14:36:31 2020 kern.info kernel: [79691.969623] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.970271] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x2f4/0x300
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.970997] Modules linked in: fast_classifier xt_FULLCONENAT xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack ipt_REJECT xt_time xt_tcpudp xt_tcpmss xt_statistic xt_socket xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TPROXY xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbnet ts_kmp ts_fsm ts_bm slhc sch_cake r8152 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.971201]  asn1_decoder tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic ecb authenc arc4
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.984356] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.69 #0
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.984883] Hardware name: FriendlyElec NanoPi R2S (DT)
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.985355] pstate: 40000005 (nZcv daif -PAN -UAO)
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.985789] pc : dev_watchdog+0x2f4/0x300
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.986155] lr : dev_watchdog+0x2f4/0x300
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.986516] sp : ffff800010013da0
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.986815] x29: ffff800010013da0 x28: ffff00003ca73e80
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.987293] x27: 0000000000000004 x26: 0000000000000140
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.987773] x25: 00000000ffffffff x24: 0000000000000002
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.988252] x23: ffff00003b87a45c x22: ffff00003b87a000
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.988732] x21: ffff00003b87a480 x20: ffff800010b86000
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.989211] x19: 0000000000000000 x18: 0000000000000000
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.989688] x17: 0000000000000000 x16: 0000000000000000
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.990167] x15: 0000000000000000 x14: ffff800010c0dfc2
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.990643] x13: 0000000000000000 x12: ffff800010c0d000
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.991123] x11: ffff800010b99000 x10: 0000000000000010
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.991602] x9 : 0000000000000000 x8 : 656d697420302065
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.992082] x7 : 756575712074696d x6 : 000000000000010a
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.992560] x5 : 0000000000000001 x4 : ffff00003f59c708
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.993039] x3 : 0000000000000006 x2 : 0000000000000007
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.993518] x1 : ffff00003d8d8b80 x0 : 0000000000000039
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.993999] Call trace:
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.994233]  dev_watchdog+0x2f4/0x300
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.994575]  call_timer_fn.isra.35+0x20/0x78
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.994965]  run_timer_softirq+0x378/0x388
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.995339]  __do_softirq+0x124/0x260
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.995677]  irq_exit+0xb8/0xc8
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.995972]  handle_IPI+0x114/0x1a8
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.996288]  gic_handle_irq+0x9c/0xa8
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.996623]  el1_irq+0xb8/0x180
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.996916]  cpuidle_enter_state+0x8c/0x358
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.997294]  cpuidle_enter+0x34/0x48
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.997623]  call_cpuidle+0x1c/0x40
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.997942]  do_idle+0x1c0/0x218
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.998239]  cpu_startup_entry+0x24/0x78
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.998598]  secondary_start_kernel+0x164/0x178
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.999007] ---[ end trace 1b035ec46ce2b34c ]---
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79691.999579] r8152 4-1:1.0 eth1: Tx timeout
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79692.002880] r8152 4-1:1.0 eth1: Tx status -2
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79692.003362] r8152 4-1:1.0 eth1: Tx status -2
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79692.003824] r8152 4-1:1.0 eth1: Tx status -2
Sun Oct 25 14:36:31 2020 kern.warn kernel: [79692.004285] r8152 4-1:1.0 eth1: Tx status -2
Sun Oct 25 14:36:31 2020 kern.info kernel: [79692.682021] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sun Oct 25 14:36:31 2020 kern.err kernel: [79692.714241] r8152 4-1:1.0 eth1: Invalid ether addr 00:00:00:00:00:00
Sun Oct 25 14:36:31 2020 kern.info kernel: [79692.714841] r8152 4-1:1.0 eth1: Random ether addr 16:ff:7e:f4:4c:50
Sun Oct 25 14:36:31 2020 kern.notice kernel: [79692.717812] r8152 4-1:1.0 eth1: Promiscuous mode enabled

hyird, 2020-10-26:
万物皆断流，
自己连接数炸了而已 
关闭 sfe flow 尤其是nat1即可

aueu, 2020-10-27:
不要使用sfe
自从将sfe移出固件后，再也没断流过，我开了fullcone nat

ostrichii, 2020-10-27:
> 万物皆断流，
> 自己连接数炸了而已
> 关闭 sfe flow 尤其是nat1即可

擦，还有这坑，X86一样的配置毛线问题没有

ostrichii, 2020-10-27:
> 不要使用sfe
> 自从将sfe移出固件后，再也没断流过，我开了fullcone nat

我试试

aueu, 2020-10-27:
> > 不要使用sfe
> > 自从将sfe移出固件后，再也没断流过，我开了fullcone nat
> 
> 我试试

你使用的是本项目的r2s么
我用的上游是openwrt，然后从本项目摘取一些patch，插件取本项目以及隔壁的

以前使用其他人编译的固件，网卡用2.x版本，开了sfe，一天总有那么几下网卡重置。后来降级网卡驱动，还是偶尔网卡重置，彻底移出sfe后，就再也没发生过网卡重置的事

ostrichii, 2020-10-28:
> > > 不要使用sfe
> > > 自从将sfe移出固件后，再也没断流过，我开了fullcone nat
> > 
> > 
> > 我试试
> 
> 你使用的是本项目的r2s么
> 我用的上游是openwrt，然后从本项目摘取一些patch，插件取本项目以及隔壁的
> 
> 以前使用其他人编译的固件，网卡用2.x版本，开了sfe，一天总有那么几下网卡重置。后来降级网卡驱动，还是偶尔网卡重置，彻底移出sfe后，就再也没发生过网卡重置的事
> 现在就用openwrt上游的flow offloading了，不折腾了。
> 
> 另外，硬件问题，建议贴内核日志...

是用的本项目的R2S，贴的这段日志是内核日志插到系统日志里来的，我看了是一样的，所以就直接贴上来了
另外，请问你是用了flow offloading就没这问题了？

aueu, 2020-10-28:
> 是用的本项目的R2S，贴的这段日志是内核日志插到系统日志里来的，我看了是一样的，所以就直接贴上来了
> 另外，请问你是用了flow offloading就没这问题了？

对的，我用了 flow offloading 与 fullcone nat，wan lan 口没有如同其他人那样交换。

closed 2021-06-20 by coolsnowwolf

VwEl, 2022-10-20:
> 对的，我用了 flow offloading 与 fullcone nat，wan lan 口没有如同其他人那样交换。

 Software Offload要不要关哦，哥