coolsnowwolf/lede #5718: 云编译过程中的编译错误

Derek-G, 2020-10-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

通过P3TERX/Actions-OpenWrt 项目进行云编译过程中workflow的compile the firmware 过程中出现编译报错

### (2) 路由器型号和固件版本
软路由，coolsnowwolf/lede master分支下当前固件

### (3) 详细日志
package/Makefile:65: recipe for target 'package/install' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:108: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.

Derek-G, 2020-10-26:
转为虚拟机编译后同样编译失败，log如下：
Traceback (most recent call last):
  File "./configure", line 15, in <module>
    from distutils.spawn import find_executable as which
ModuleNotFoundError: No module named 'distutils.spawn'
Makefile:154: recipe for target '/home/meisgt/lede/build_dir/target-x86_64_musl/node-v12.16.1/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/meisgt/lede/build_dir/target-x86_64_musl/node-v12.16.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/meisgt/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#3.23#1.94#5.01
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/meisgt/lede'
package/Makefile:107: recipe for target '/home/meisgt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/meisgt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/meisgt/lede'
/home/meisgt/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

bin20088, 2020-10-26:
同样云编译，天天编译也没报错

Dimitri2020007, 2020-10-26:
from distutils.spawn import find_executable as which
ModuleNotFoundError: No module named 'distutils.spawn'

这个模块没有的

Derek-G, 2020-10-26:
> from distutils.spawn import find_executable as which
> ModuleNotFoundError: No module named 'distutils.spawn'
> 
> 这个模块没有的

sudo apt install python3-distutils -y
我尝试这么去解决，还没有得到最新的变异结果。好像是node部分的原因，google一圈在node project 里面issues看到的方法，不知道能不能好用。

Dimitri2020007, 2020-10-28:
改过action脚本了吗？

fonlan, 2020-10-29:
我这边执行sudo apt-get install libtinfo5后解决了，你们也可以试试是不是同一个原因

Dimitri2020007, 2020-10-30:
脚本或者实体编译可以添加这个软件libtinfo5

lingan1996, 2020-11-05:
> 我这边执行sudo apt-get install libtinfo5后解决了，你们也可以试试是不是同一个原因

我安装libtinfo5后还是编译失败，报错和之前的一样的。

Dimitri2020007, 2020-11-07:
把action源码上传到fo过的lede目录底下。
删除.github/workflow中的云编译脚本。（在这之前先把action workflow文件复制进去再删就不会连目录一起删了）

fo源码根目录上传diy-part1.sh和diy-part2.sh

虚拟机克隆lede源码，更新安装feeds后直接make menuconfig

选择自己想要的软件，保存menuconfig文件。

在fo源码根目录下创建.config文件

sftp连接到虚拟机，找到lede目录下.config，复制所有内容到新建的.config文件里

action开始编译就行了。

直接使用action源码编译是会出错的。

Derek-G, 2020-11-20:
已经找到问题原因， 是因为同时选择了两个版本的smb所导致，谢谢大家的解答！希望能帮助到其他的小伙伴！！

Derek-G, 2020-11-20:
> 把action源码上传到fo过的lede目录底下。
> 删除.github/workflow中的云编译脚本。（在这之前先把action workflow文件复制进去再删就不会连目录一起删了）
> 
> fo源码根目录上传diy-part1.sh和diy-part2.sh
> 
> 虚拟机克隆lede源码，更新安装feeds后直接make menuconfig，但不执行下载软件包和编译过程。
> 
> 选择自己想要的软件，保存menuconfig文件。
> 
> 在fo源码根目录下创建.config文件
> 
> sftp连接到虚拟机，找到lede目录下.config，复制所有内容到新建的.config文件里
> 
> action开始编译就行了。
> 
> 直接使用action源码编译是会出错的。

我就是用的这个方法进行云编译的，步骤没有问题。但还是要感谢你的回答，希望能帮助到其他人。

closed 2020-11-20 by Derek-G