coolsnowwolf/lede #5719: Action 编译错误 （10月 V2RAY 升级后）

yfdoor, 2020-10-26:
自用一直采用 OP 19.07.04 + Lean package 的方式通过Action 编译，之前一直正常，最近发现从 10月1 之后，编译开始报错了，查看错误代码并在 issue 区查找发现，似乎是和 V2RAY 升级有关，我看有人说需要 feeds update，可是通过 action 编译每次都是自动执行 feeds update 的啊，而且 feeds update 这步也没有任何的报错信息。
请问下各位，应该如何解决呢？万分感谢. 

`2020-10-25T01:35:01.6590376Z make[3]: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/v2ray-core-4.31.2/.built] Error 2

2020-10-25T01:35:01.6594163Z Makefile:308: recipe for target '/home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/v2ray-core-4.31.2/.built' failed

2020-10-25T01:35:01.6595119Z make[3]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/package/lean/v2ray'

2020-10-25T01:35:01.6597522Z make[2]: *** [package/lean/v2ray/compile] Error 2
2020-10-25T01:35:01.6598091Z time: package/lean/v2ray/compile#5.46#1.11#14.15
2020-10-25T01:35:01.6598977Z package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
2020-10-25T01:35:01.6618932Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
2020-10-25T01:35:01.6668151Z make[1]: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2

2020-10-25T01:35:01.6669468Z package/Makefile:107: recipe for target '/home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed

2020-10-25T01:35:01.6670525Z make: *** [world] Error 2`

[Action Link](https://github.com/yfdoor/OpenWrt/actions)

Dimitri2020007, 2020-10-26:
是不是跟naiveproxy一样过了版本没过补丁啊？

bin20088, 2020-10-26:
同样云编译，天天编译也没报错

Ljzkirito, 2020-10-26:
我过去看了看[log](https://github.com/yfdoor/OpenWrt/runs/1303506459?check_suite_focus=true)
问题在这
`32428 note: module requires Go 1.14`
v2ray新版要求Go 1.15+，更新packages/lang/golang试试

yfdoor, 2020-10-26:
> 我过去看了看[log](https://github.com/yfdoor/OpenWrt/runs/1303506459?check_suite_focus=true)
> 问题在这
> `32428 note: module requires Go 1.14`
> v2ray新版要求Go 1.15+，更新packages/lang/golang试试
> v2ray一直要求使用最新的Golang，编译不通过一般是Golang版本低

看了一下 lean 里的 feed package golong 是 1.15.2
OP 官方的 19.07.4 package golong 是 1.13.15， master 分支已经是 1.15.3，似乎是和这个有关，请问下在 不修改 feeds.conf.default package 路径情况下，在 action 中 有办法只更新 golong 到 新版吗？

Ljzkirito, 2020-10-26:
> 请问下在 不修改 feeds.conf.default package 路径情况下，在 action 中 有办法只更新 golong 到 新版吗？

仅供参考
`./scripts/feeds update -a`后执行
```pushd feeds/packages/lang
rm -fr golang
svn co https://github.com/coolsnowwolf/packages/trunk/lang/golang
popd
```

alienaltman, 2020-10-27:
> 自用一直采用 OP 19.07.04 + Lean package 的方式通过Action 编译，之前一直正常，最近发现从 10月1 之后，编译开始报错了，查看错误代码并在 issue 区查找发现，似乎是和 V2RAY 升级有关，我看有人说需要 feeds update，可是通过 action 编译每次都是自动执行 feeds update 的啊，而且 feeds update [...]

请问这个解决了吗？怎么解决的？我还是一样错误。

yfdoor, 2020-10-31:
> 请问这个解决了吗？怎么解决的？我还是一样错误。

就是 golong 版本的问题，请参考上面 Ljzkirito 的方法，修改package 或者直接使用 master 分支来编译即可。

closed 2020-10-31 by yfdoor