coolsnowwolf/lede #5722: Existing conffile lede/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at lede/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.

zhaochengwei, 2020-10-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： * resolve_conffiles: Existing conffile /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
 * check_data_file_clashes: Package libgd wants to install file /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so

### (2) 路由器型号和固件版本

A：x86 软路由

### (3) 详细日志

A：Collected errors:
 * resolve_conffiles: Existing conffile /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
 * check_data_file_clashes: Package libgd wants to install file /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so
	But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3
	But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3.0.8
	But that file is already provided by package  * libgd-full
 * opkg_install_cmd: Cannot install package luci-app-vnstat.
 * check_data_file_clashes: Package libgd wants to install file /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so
	But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3
	But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/douding/develop/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3.0.8
	But that file is already provided by package  * libgd-full
 * opkg_install_cmd: Cannot install package libgd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/douding/develop/lede'
package/Makefile:108: recipe for target '/home/douding/develop/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/douding/develop/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/douding/develop/lede'
Build failed - please re-run with -j1 to see the real error message
/home/douding/develop/lede/include/toplevel.mk:222: recipe for target 'world' failed

hxlls, 2020-10-26:
编译命令如下:
首先装好 Ubuntu 64bit，推荐 Ubuntu 18 LTS x64

命令行输入 sudo apt-get update ，然后输入 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync

使用 git clone https://github.com/coolsnowwolf/lede 命令下载好源代码，然后 cd lede 进入目录

./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make -j8 download V=s 下载dl库（国内请尽量全局科学上网）

输入 make -j1 V=s （-j1 后面是线程数。第一次编译推荐用单线程）即可开始编译你要的固件了。

请尊重作者的劳动，看准看完并按照执行

closed 2020-10-27 by coolsnowwolf

zhaochengwei, 2021-01-07:
> 编译命令如下:
> 首先装好 Ubuntu 64bit，推荐 Ubuntu 18 LTS x64
>
> 命令行输入 sudo apt-get update ，然后输入 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch [...]

我仔细执行了噢，但是没有相应的插件和主题，需要额外的配置才会有。

zhaochengwei, 2021-01-07:
[config.txt](https://github.com/coolsnowwolf/lede/files/5782420/config.txt)
贴上我的config，开了大部分选项，可以编译通过。有问题欢迎留言~