coolsnowwolf/lede #5724: feed install 的时候一大片警告闪过去了，强迫症犯了

HashimotoSakura, 2020-10-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：拉代码update feeds后install，一大片黄字警告闪过去了，请问对固件完整性影响大吗？

### (2) 路由器型号和固件版本

A：newifi d2

### (3) 详细日志

A：WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'antfs-mount', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libgstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1-libs', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/libs/libselinux/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a dependency on 'ksmbd-server', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cpufreq/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'smartmontools', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'btrfs-progs', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a dependency on 'docker-ce', which does not exist
[... 104 lines cut ...]
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'attr', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libaio', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libavahi-client', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'python3-cryptodome', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libarchive', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'acl', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'attr', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'libtasn1/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'rpcsvc-proto/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'python3/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'perl/host', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a build dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/tcpping/Makefile' has a dependency on 'libnet-1.2.x', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-system', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/v2ray-plugin/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

Ljzkirito, 2020-10-28:
你再feed install一次看看，还有没有WARNING
首次feed install都会这样

HashimotoSakura, 2020-10-28:
再次就没有了，好奇怪

jjit, 2020-10-28:
因为feeds update，会根据feeds.conf.default文件，下载依赖包到feeds文件夹
还没有放到 package文件夹里，第一次feeds install会先扫描package目录
然后再把 feeds文件夹里的依赖包，通过软连接方式放进 package目录
所以原先已经在 package目录里的插件，第一次会报 缺少一些依赖包
之后再运行 就不会报了，因为 feeds文件夹已经放在 package目录里了

HashimotoSakura, 2020-10-29:
谢谢，了解了

closed 2020-10-29 by HashimotoSakura