coolsnowwolf/lede #5727: 最新版本v2ray服务端无法启动

Dimitri2020007, 2020-10-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用github action在线编译，luci接口部署在nginx上，已安装uwsgi所有插件。删除后重建依然无法启动。

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

Thu Oct 29 19:40:06 2020 daemon.err nginx[6023]: 2020/10/29 19:40:06 [error] 6295#0: *207 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.9, server: _lan, request: "GET /cgi-bin/luci/admin/vpn/v2ray_server/get_log?_=0.8359214813875395 HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.1.5", referrer: "https://192.168.1.5/cgi-bin/luci/admin/vpn/v2ray_server"
Thu Oct 29 19:40:06 2020 daemon.info uwsgi-luci: 2020/10/29 19:40:06 mmap: cannot allocate memory
Thu Oct 29 19:40:07 2020 daemon.info uwsgi-luci: 2020/10/29 19:40:07 mmap: cannot allocate memory
Thu Oct 29 19:40:07 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:07 2020 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci/admin/vpn/v2ray_server/users_status?index=0&id=cfg022e84&_=0.7811007600172424 (ip 192.168.1.9) !!!
Thu Oct 29 19:40:07 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:07 2020 - uwsgi_response_writev_headers_and_body_do(): Broken pipe [core/writer.c line 306] during GET /cgi-bin/luci/admin/vpn/v2ray_server/users_status?index=0&id=cfg022e84&_=0.7811007600172424 (192.168.1.9)
Thu Oct 29 19:40:07 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:07 2020 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci/admin/vpn/v2ray_server/users_status?index=1&id=cfg032e84&_=0.17753530844276666 (ip 192.168.1.9) !!!
Thu Oct 29 19:40:07 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:07 2020 - uwsgi_response_writev_headers_and_body_do(): Broken pipe [core/writer.c line 306] during GET /cgi-bin/luci/admin/vpn/v2ray_server/users_status?index=1&id=cfg032e84&_=0.17753530844276666 (192.168.1.9)
Thu Oct 29 19:40:07 2020 daemon.info uwsgi-luci: invalid CGI response !!!
Thu Oct 29 19:40:07 2020 daemon.err nginx[6023]: 2020/10/29 19:40:07 [error] 6295#0: *217 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.9, server: _lan, request: "GET /cgi-bin/luci/admin/vpn/v2ray_server/get_log?_=0.7700059351075654 HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.1.5", referrer: "https://192.168.1.5/cgi-bin/luci/admin/vpn/v2ray_server"
Thu Oct 29 19:40:08 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:08 2020 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci/admin/services/shadowsocksr/ping?index=14&domain=45.79.90.66&port=8097&_=0.8884490327090584 (ip 192.168.1.9) !!!
Thu Oct 29 19:40:08 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:08 2020 - uwsgi_response_writev_headers_and_body_do(): Broken pipe [core/writer.c line 306] during GET /cgi-bin/luci/admin/services/shadowsocksr/ping?index=14&domain=45.79.90.66&port=8097&_=0.8884490327090584 (192.168.1.9)
Thu Oct 29 19:40:09 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:09 2020 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci/admin/services/shadowsocksr/ping?index=7&domain=47.245.30.243&port=2899&_=0.5317272846690897 (ip 192.168.1.9) !!!
Thu Oct 29 19:40:09 2020 daemon.info uwsgi-luci: Thu Oct 29 19:40:09 2020 - uwsgi_response_writev_headers_and_body_do(): Broken pipe [core/writer.c line 306] during GET /cgi-bin/luci/admin/services/shadowsocksr/ping?index=7&domain=47.245.30.243&port=2899&_=0.5317272846690897 (192.168.1.9)
Thu Oct 29 19:40:10 2020 daemon.info uwsgi-luci: invalid CGI response !!!
Thu Oct 29 19:40:10 2020 daemon.err nginx[6023]: 2020/10/29 19:40:10 [error] 6295#0: *207 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.9, server: _lan, request: "GET /cgi-bin/luci/admin/vpn/v2ray_server/get_log?_=0.9044438760654794 HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.1.5", referrer: "https://192.168.1.5/cgi-bin/luci/admin/vpn/v2ray_server"
Thu Oct 29 19:40:10 2020 daemon.info uwsgi-luci: 2020/10/29 19:40:10 mmap: cannot allocate memory
Thu Oct 29 19:40:10 2020 daemon.info uwsgi-luci: 2020/10/29 19:40:10 mmap: cannot allocate memory
Thu Oct 29 19:40:10 2020 daemon.info uwsgi-luci: invalid CGI response !!!
Thu Oct 29 19:40:10 2020 daemon.err nginx[6023]: 2020/10/29 19:40:10 [error] 6295#0: *217 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.9, server: _lan, request: "GET /cgi-bin/luci/admin/vpn/v2ray_server/get_log?_=0.7222404367905082 HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.1.5", referrer: "https://192.168.1.5/cgi-bin/luci/admin/vpn/v2ray_server"
Thu Oct 29 19:40:13 2020 daemon.info uwsgi-luci: invalid CGI response !!!
Thu Oct 29 19:40:13 2020 daemon.err nginx[6023]: 2020/10/29 19:40:13 [error] 6295#0: *228 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.9, server: _lan, request: "GET /cgi-bin/luci/admin/vpn/v2ray_server/get_log?_=0.4796938374391959 HTTP/1.1", upstream: "uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.1.5", referrer: "https://192.168.1.5/cgi-bin/luci/admin/vpn/v2ray_server"
Thu Oct 29 19:40:14 2020 daemon.info uwsgi-luci: 2020/10/29 19:40:14 mmap: cannot allocate memory
Thu Oct 29 19:40:14 2020 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.
Thu Oct 29 19:40:18 2020 daemon.info uwsgi-luci: uWSGI worker 2 cheaped.
Thu Oct 29 19:40:56 2020 daemon.info uwsgi-luci: 2020/10/29 19:40:56 mmap: cannot allocate memory

RockJohnson503, 2020-10-30:
trojan-go一样的，你得进shell重启，页面调用会报错 mmap: cannot allocate memory

RockJohnson503, 2020-10-30:
#5692 解决了

Dimitri2020007, 2020-10-30:
好吧我试试

closed 2020-10-30 by Dimitri2020007