coolsnowwolf/lede #5728: 启动问题：Wrong EFI loader signature

seasun2k, 2020-10-30:
1. 编译的固件启动显示Wrong EFI loader signature，没用EFI...如何解决？
2. 启动信息显示booting the kernel后就出现plase press Enter to activate this console.没有出现之前的启动信息，如何恢复显示之前的启动信息控制台显示？

Dimitri2020007, 2020-10-30:
配置menuconfig>target images>选中use console terminal（in addition to serial）选项就可以恢复启动信息显示。

seasun2k, 2020-10-30:
> 配置menuconfig>target images>选中use console terminal（in addition to serial）选项就可以恢复启动信息显示。

解决了...多谢！ 一直以为这个选项是专门用于串口的呢.....:(

> 禁用安全启动并清空安全启动密钥，启动中有legacy boot或csm的打开就可以。

我用的软路由，openwrt装在ESXi里，虚拟机里没有安全启动选项，选择的也是BOIS启动....我试了几个ROM，也都有这个错误提示。：（

Dimitri2020007, 2020-10-30:
你转盘的时候是不是错把efi系统的固件强装到esxi里了？

seasun2k, 2020-11-03:
> 你转盘的时候是不是错把efi系统的固件强装到esxi里了？
> 普通bios启动的镜像后面是没有efi的。
> 
> openwrt-x86-64-generic-squashfs-combined-efi.img（efi用的）
> openwrt-x86-64-generic-squashfs-combined.img（bios用的）

确定是没有efi的，试了几个固件，都会出现Wrong EFI loader signature。使用中也没有什么影响.

Dimitri2020007, 2020-11-03:
你试试虚拟机uefi启动。若没有问题就行了，关闭安全启动选项只有在选了uefi以后才出来的，普通bios是没有安全启动的

closed 2021-06-20 by coolsnowwolf