coolsnowwolf/lede #5731: 转盘后在esxi里添加现有硬盘不能识别出来文件

Eintler, 2020-10-31:
自己编译完后
用StarWind V2V Converter转盘openwrt-x86-64-generic-squashfs-combined.img
生成openwrt-x86-64-generic-squashfs-combined.vmdk 和 openwrt-x86-64-generic-squashfs-combined-flat.vmdk
也已经上传成功。
但是在添加现有硬盘时，却显示不出来任何文件。

求解啊。

Dimitri2020007, 2020-10-31:
我也碰到这个问题了

eramicro, 2020-10-31:
建议你们在配置里勾选vmdk的选项，自动生成esxi硬盘文件。

Eintler, 2020-10-31:
@eramicro 能详细说一下在哪里勾选嘛？现在还是不太熟。

jjit, 2020-10-31:
在 Target Images 里面，Build VMware image files (VMDK)

hxlls, 2020-10-31:
生成的vmdk 是VMware workstion 的磁盘文件不是Esxi 磁盘文件，自己用IMG转盘完事

Dimitri2020007, 2020-10-31:
menuconfig>target images>中选的vmdk只能用在vmware workstation里。esxi用不了

Eintler, 2020-10-31:
我就是用这种方式，转盘后，在添加现有硬盘时找不到vmdk的啊。

hxlls, 2020-10-31:
转后生成两个文件，其中一个是-fat的，需要两个都上传后才会自动合成标准ESXI使用的磁盘文件

Eintler, 2020-10-31:
是的，就是2个
一个是openwrt-x86-64-generic-squashfs-combined.vmdk
另一个是openwrt-x86-64-generic-squashfs-combined-flat.vmdk
在数据存储浏览器可以看到，唯独在添加现有硬盘时就看不到了。

Eintler, 2020-10-31:
解决了，原来是手欠把文件名自带的后缀名（1）给删了，估计是校验没通过。谢谢大家的帮助。

eramicro, 2020-10-31:
> 
> 
> menuconfig>target images>中选的vmdk只能用在vmware workstation里。esxi用不了

我非常确定是可以使用的，用不了你得看清楚提示，不能使用是因为你没改磁盘控制器位置（IDE控制器0），没有争议，动手试试你就清楚了！！！

eramicro, 2020-10-31:
> 
> 
> 生成的vmdk 是VMware workstion 的磁盘文件不是Esxi 磁盘文件，自己用IMG转盘完事

你自己亲身实践下就啥都清楚了。

Dimitri2020007, 2020-11-04:
使用qemu-img convert 执行img转盘到vhdx

qemu-img convert -f raw -O vmdk <源>.img <目标>.vmdk

qemu-img check <目标>.vmdk （不出意外肯定是no error）

转盘完成在esxi添加硬盘的时候把硬盘控制器从scsi/sata全改到ide/主要

esxi 6.7测试成功可以正常使用。

closed 2021-06-20 by coolsnowwolf