coolsnowwolf/lede #5733: 有人能看一下这个错误吗？启动后老刷屏，偶尔会重启。

Dimitri2020007, 2020-10-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：这个错误会在固件启动之后疯狂的刷屏，而且一直在重复刷这个。所用固件是efi版本，使用qemu-img工具转盘img为vhdx在hyper-v使用，hyper-v虚拟机，2核，4g内存。

esxi转盘无法识别，跳过测试，pve虚拟机未测试无法提供详细信息。

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：后台一直不停刷这个，不知道到啥时候自己就停下了。

[ 7864.412335] ------------[ cut here ]------------
[ 7864.416235] WARNING: CPU: 0 PID: 13220 at 0xffffffffa162d22a
[ 7864.416235] Modules linked in: fast_classifier ath9k iwl4965 iwl3945 ath9k_htc ath9k_common wl18xx wl12xx snd_hda_codec_conexant snd_hda_codec_cmedia snd_hda_codec_cirrus snd_hda_codec_ca0110 snd_hda_codec_analog rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci rt2800usb rt2800pci rt2800mmio rt2800lib rt2500usb rt2500pci rt2400pci qcserial pppoe ppp_async option mt76x0u mt76x0e mt76x0_common iwlmvm iwlegacy iwldvm ipw2200 ipw2100 ipw cdc_mbim carl9170 btcoexist brcmfmac b43legacy b43 ath9k_hw ath6kl_usb ath6kl_sdio ath6kl_core ath5k ath10k_pci ath10k_core ath adm8211 zd1211rw wlcore usb_wwan usb_serial_simple ueagle_atm ti_usb_3410_5052 sr9700 speedtch snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_si3054 snd_hda_codec_realtek snd_hda_codec_idt snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_codec_ca0132 snd_hda_codec smsc95xx sierra_net sierra rtw88_pci rtw88_core
[ 7864.416235]  rtlwifi rtl8xxxu rtl8812au rtl818x_pci rtl8187 rt73usb rt61pci rt2x00usb rt2x00pci rt2x00mmio rt2x00lib rsi_usb rsi_sdio rsi_91x rndis_host r8723bs qmi_wwan pppox ppp_mppe ppp_generic plusb pl2303 p54usb p54pci p54common oti6858 orinoco_plx orinoco_pci orinoco_cs orinoco nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet mwl8k mwifiex_sdio mwifiex mt7915e mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663s mt7663_usb_sdio_common mt7615e mt7615_common mt7603e mt7601u mt76_usb mt76_sdio mt76 mos7840 mos7720 mct_u232 mcs7830 mac80211_hwsim mac80211 libipw libertas_sdio libertas keyspan kalmia iwlwifi iscsi_tcp io_edgeport huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 cxacru cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a ath9k_pci_owl_loader ath3k asix ark3116 amdgpu zl6100 yealink xt_u32 xt_time xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent
[ 7864.506260]  xt_quota2 xt_quota xt_psd xt_policy xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_lscan xt_limit xt_length2 xt_length xt_ipv4options xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_geoip xt_fuzzy xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connlabel xt_connbytes xt_condition xt_comment xt_cluster xt_bpf xt_addrtype xt_TRACE xt_TPROXY xt_TEE xt_TCPMSS xt_SYSRQ xt_REDIRECT xt_PROTO xt_NFQUEUE xt_NFLOG xt_NETMAP xt_MASQUERADE xt_LUA xt_LOGMARK xt_LOG xt_LED xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_DNETMAP xt_DHCPMAC xt_CT xt_CLASSIFY xt_CHECKSUM xt_DELUDE xt_TARPIT ipt_REJECT xt_tcpudp xt_CHAOS xt_ACCOUNT xr_usb_serial_common wlcore_sdio wireguard w83793 visor via_velocity via_rhine vhci_hcd usbserial usbnet usbmon usblp usbip_host usbip_core usbatm ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda uinput tulip ts_fsm ts_bm tmp103 tmp102 snd_usbmidi_lib
[ 7864.705616]  snd_intel_nhlt snd_hda_core slhc sky2 skge sis900 sis190 sht21 sch_cake rtl8150 rtc_pcf8563 rtc_pcf2127 rtc_pcf2123 rtc_isl1208 rtc_em3027 rtc_ds1672 rtc_ds1374 rtc_ds1307 rotary_encoder rfcomm regmap_spi radeon r8712u(C) r8169 r8168 r8152 r8125 r6040 pmbus_core pegasus pcnet32 niu nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nfnetlink_queue nfnetlink_log nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_dup_netdev nf_dup_ipv6 nf_dup_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache
[ 7864.705616]  nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount netatop ne2k_pci mmc_spi mlx5_core mlx4_en mlx4_core mdio_gpio mdio_bitbang mcp3021 matrix_keymap macvlan lzo ltc4151 lm95241 lm92 lm90 lm85 lm77 lm75 lm63 libiscsi_tcp libiscsi lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 kvm ksmbd kaweth joydev irqbypass iptable_raw iptable_nat iptable_mangle iptable_filter ipt_rpfilter ipt_ah ipt_ECN ipt_CLUSTERIP ipheth ip6table_raw ip6t_rpfilter ip_tables ina2xx ina209 iavf i40e hso hidp hci_uart gpu_sched forcedeth ezusb exfat ethoc et131x ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_nflog ebt_mark_m ebt_mark ebt_log ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 e1000e e100 dmx_usb dme1737 crc7 crc_ccitt cordic compat_xtables compat cm109 chaoskey cdc_wdm cdc_acm btusb btmrvl_sdio btmrvl btintel brcmutil br_netfilter
[ 7864.705616]  bnx2x bnx2 bnep bluetooth_6lowpan bluetooth be2net atl2 atl1e atl1c atl1 at25 at24 asn1_decoder arptable_filter arpt_mangle arp_tables alx adt7475 ads7846 8390 8250_exar 8139too 8139cp fuse tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred configs 6lowpan sg evdev tmp421 drivetemp adt7410 adt7x10 adcxx ad7418 snd_soc_ac97 snd_soc_core gpio_pcf857x gpio_pca953x lp parport_pc i2c_dev ledtrig_usbport spi_ks8995 siit ppdev parport snd_pcsp gpio_beeper echo cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink st rtl8366s rtl8366rb rtl8306 ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh
[ 7864.705616]  ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah mvsw61xx ip17xx b53_mdio b53_common nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 pinctrl_mcp23s08 regmap_i2c regmap_core nfsd nfs msdos bonding 3c59x snd_via82xx snd_intel8x0 snd_ens1371 snd_seq_midi snd_seq_midi_event snd_seq snd_mpu401_uart ixgbevf ixgbe igbvf igb e1000 snd_ac97_codec ac97_bus ifb dummy ip6_udp_tunnel udp_tunnel snd_dummy mdio ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ipip tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti rpcsec_gss_krb5 auth_rpcgss oid_registry dm_raid veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore nbd mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat udf crc_itu_t ntfs lockd sunrpc grace minix hfsplus hfs cramfs configfs cifs autofs4 rxrpc 9p dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax br2684
[ 7865.593175]  atm aoe multipath fscache bcma 9pnet_virtio 9pnet raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 linear md_mod pcmcia pcmcia_core nls_utf8 nls_koi8_r nls_cp1255 nls_iso8859_6 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp950 nls_cp936 nls_cp932 nls_cp866 nls_cp864 nls_cp862 nls_cp852 nls_cp850 nls_cp775 nls_cp437 nls_cp1251 nls_cp1250 zram zsmalloc natsemi eeprom_93cx6 bfq ecdh_generic ecc shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_ssse3 sha512_generic sha256_ssse3 sha256_generic libsha256 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg pcbc michael_mic md5 md4 kpp hmac ghash_generic gcm fcrypt echainiv des_generic libdes deflate cts ctr cmac ccm cbc authenc arc4 crypto_acompress uas sdhci_pltfm input_polldev xhci_plat_hcd dwc3 dwc2 softdog rtc_s35390a rtc_rx8025 rtc_rs5c372 iTCO_wdt iTCO_vendor_support rtl8367b swconfig rtl8366_smi sata_via sata_sil24 sata_sil
[... cut ...]
[ 7867.707282] RSP: 002b:00007fe18644b788 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[ 7867.707282] RAX: ffffffffffffffda RBX: 00007fe18644cb20 RCX: 00007fe186d08373
[ 7867.707282] RDX: 000000000000001e RSI: 00007fe18644b800 RDI: 000000000000002b
[ 7867.707282] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
[ 7867.707282] R10: 0000000000000000 R11: 0000000000000246 R12: 000000000000002b
[ 7867.707282] R13: 0000000000000002 R14: 0000000000000030 R15: 0000000000000003
[ 7867.707282] ---[ end trace d4ba6897653de650 ]---

closed 2020-11-08 by Dimitri2020007

galaxyskyknight, 2020-12-27:
@Dimitri2020007 请问你这个问题解决了吗，我遇到和你一样的问题，#6114

Dimitri2020007, 2020-12-27:
我已经解决了，编译的时候sfe和flowoffload两个只需要选一个就可以避免这类问题。