coolsnowwolf/lede #5735: esxi7.0u1 +thinkcenter M73 单网口 不定时挂起

kalagxw, 2020-11-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述
wan：虚拟交换机1+e1000e+出口物理网卡i217-v（管理口也在此虚拟交换，单网卡似乎不能直通）
lan：虚拟交换机2+e1000e+下挂ubuntu
### (1) 具体问题

A：能启动正常使用，一段时间后（时长不确定），cpu100% ，网络无法访问。重启可以恢复

### (2) 路由器型号和固件版本

A：试了不同版本内核X86 +sfe 开启docker

### (3) 详细日志

A： rcu: INFO: rcu_sched self-detected stall on CPU
[42586.865142] rcu: 	1-....: (1 GPs behind) idle=c3e/0/0x3 softirq=1482648/1482648 fqs=363055 
[42586.865229] rcu: 	 (t=726121 jiffies g=2764357 q=282156)
[42586.865293] NMI backtrace for cpu 1
[42586.865332] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G        W         4.19.123 #0
[42586.865406] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 05/28/2020
[42586.865509] Call Trace:
[42586.865538]  <IRQ>
[42586.865582]  dump_stack+0x6d/0x95
[42586.865620]  nmi_cpu_backtrace+0x8a/0xa0
[42586.865674]  ? lapic_can_unplug_cpu+0xa0/0xa0
[42586.865718]  nmi_trigger_cpumask_backtrace+0x51/0xc0
[42586.865768]  arch_trigger_cpumask_backtrace+0x14/0x20
[42586.865828]  rcu_dump_cpu_stacks+0x82/0xca
[42586.865870]  rcu_check_callbacks+0x33a/0x6a0
[42586.865918]  ? account_system_time+0x56/0x60
[42586.865966]  ? tick_init_highres+0x20/0x20
[42586.866010]  update_process_times+0x2a/0x50
[42586.866053]  tick_sched_timer+0x144/0x180
[42586.866094]  ? tick_init_highres+0x20/0x20
[42586.866135]  __hrtimer_run_queues+0x8f/0x140
[42586.866179]  hrtimer_interrupt+0xea/0x200
[42586.866224]  smp_apic_timer_interrupt+0x76/0x90
[42586.866270]  apic_timer_interrupt+0xf/0x20
[42586.866315] RIP: 0010:native_queued_spin_lock_slowpath+0x176/0x190
[42586.866377] Code: 01 00 00 00 f0 0f b1 0f 74 15 eb e9 48 8b 32 48 85 f6 75 04 f3 90 eb f4 c7 46 08 01 00 00 00 65 ff 0d 4a ea f5 7e eb 10 f3 90 <8b> 07 85 c0 75 f8 f0 0f b1 17 85 c0 75 f2 5d c3 66 2e 0f 1f 84 00
[42586.866598] RSP: 0018:ffff88803eb03af0 EFLAGS: 00000202 ORIG_RAX: ffffffffffffff13
[42586.866674] RAX: 0000000000007f9d RBX: ffff88802d42bb40 RCX: 0000000060cec985
[42586.866754] RDX: 0000000000000001 RSI: 0000000000007f9d RDI: ffff88802d42bb44
[42586.866824] RBP: ffff88803eb03af0 R08: 0000000000000000 R09: 0000000000000001
[42586.866894] R10: 0000000000000001 R11: 0000000060cec985 R12: ffff88802d42bb44
[42586.866964] R13: 0000000000000001 R14: ffff88803eb03b48 R15: ffff88802cac2c40
[42586.867034]  ? apic_timer_interrupt+0xa/0x20
[42586.867082]  _raw_spin_lock+0x1a/0x20
[42586.867130]  fast_classifier_recv+0x1798/0x1def [fast_classifier]
[42586.867196]  ? br_pass_frame_up+0x10f/0x120
[42586.867240]  ? nf_nat_inet_fn+0x181/0x280 [nf_nat]
[42586.867288]  fast_classifier_recv+0x1de1/0x1def [fast_classifier]
[42586.867359]  nf_hook_slow+0x36/0xa0
[42586.867398]  ip_output+0x7a/0xa0
[42586.867432]  ? __ip_flush_pending_frames.isra.50+0x80/0x80
[42586.867487]  ip_forward_finish+0x42/0x50
[42586.867526]  ip_forward+0x36d/0x3e0
[42586.867562]  ? ip4_key_hashfn+0xb0/0xb0
[42586.867601]  ip_rcv_finish+0x3a/0x40
[42586.867638]  ip_rcv+0x83/0x90
[42586.867669]  ? ip_rcv_finish_core.isra.13+0x250/0x250
[42586.867727]  __netif_receive_skb_one_core+0x43/0x50
[42586.867777]  __netif_receive_skb+0x54/0x60
[42586.867818]  process_backlog+0xad/0x150
[42586.867857]  net_rx_action+0xf3/0x270
[42586.867895]  __do_softirq+0xba/0x1f4
[42586.867940]  irq_exit+0x62/0x90
[42586.867972]  do_IRQ+0xaf/0xd0
[42586.868003]  common_interrupt+0xf/0xf
[42586.868040]  </IRQ>
[42586.868063] RIP: 0010:native_safe_halt+0x17/0x20
[42586.868110] Code: 07 0f 00 2d 3b 7c 58 00 f4 5d c3 0f 1f 84 00 00 00 00 00 8b 05 8a 63 b3 00 55 48 89 e5 85 c0 7e 07 0f 00 2d 1b 7c 58 00 fb f4 <5d> c3 90 90 90 90 90 90 90 55 48 89 e5 e8 57 46 a5 ff 65 8b 05 60
[42586.868290] RSP: 0018:ffffc9000007fea0 EFLAGS: 00000246 ORIG_RAX: ffffffffffffffd8
[42586.868364] RAX: 0000000000000000 RBX: 0000000000000001 RCX: ffff88803eb1a440
[42586.868434] RDX: ffffffff82032210 RSI: ffff88803eb1a440 RDI: 0000000000000000
[42586.868504] RBP: ffffc9000007fea0 R08: 0000000000000001 R09: 0000000000000001
[42586.868574] R10: 0000000000000001 R11: 0000000000000000 R12: 0000000000000000
[42586.868654] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[42586.868754]  default_idle+0x10/0x20
[42586.868794]  arch_cpu_idle+0x10/0x20
[42586.868831]  default_idle_call+0x27/0x30
[42586.868871]  do_idle+0xb3/0x180
[42586.868903]  cpu_startup_entry+0x5f/0x70
[42586.868943]  start_secondary+0x196/0x1a0
[42586.868986]  secondary_startup_64+0xa4/0xb0
[42766.894916] rcu: INFO: rcu_sched self-detected stall on CPU
[42766.894988] rcu: 	1-....: (1 GPs behind) idle=c3e/0/0x3 softirq=1482648/1482648 fqs=372057 
[42766.895080] rcu: 	 (t=744124 jiffies g=2764357 q=293047)
[42766.895140] NMI backtrace for cpu 1
[42766.895185] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G        W         4.19.123 #0
[42766.895267] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 05/28/2020
[42766.895387] Call Trace:
[42766.895420]  <IRQ>
[42766.895460]  dump_stack+0x6d/0x95
[42766.895500]  nmi_cpu_backtrace+0x8a/0xa0
[42766.895547]  ? lapic_can_unplug_cpu+0xa0/0xa0
[42766.895597]  nmi_trigger_cpumask_backtrace+0x51/0xc0
[42766.895653]  arch_trigger_cpumask_backtrace+0x14/0x20
[42766.895713]  rcu_dump_cpu_stacks+0x82/0xca
[42766.895760]  rcu_check_callbacks+0x33a/0x6a0
[42766.895811]  ? account_system_time+0x56/0x60
[42766.895861]  ? tick_init_highres+0x20/0x20
[42766.895909]  update_process_times+0x2a/0x50
[42766.895957]  tick_sched_timer+0x144/0x180
[42766.896003]  ? tick_init_highres+0x20/0x20
[42766.896050]  __hrtimer_run_queues+0x8f/0x140
[42766.896099]  hrtimer_interrupt+0xea/0x200
[42766.896146]  smp_apic_timer_interrupt+0x76/0x90
[42766.896198]  apic_timer_interrupt+0xf/0x20
[42766.896246] RIP: 0010:native_queued_spin_lock_slowpath+0x178/0x190
[42766.896319] Code: 00 00 f0 0f b1 0f 74 15 eb e9 48 8b 32 48 85 f6 75 04 f3 90 eb f4 c7 46 08 01 00 00 00 65 ff 0d 4a ea f5 7e eb 10 f3 90 8b 07 <85> c0 75 f8 f0 0f b1 17 85 c0 75 f2 5d c3 66 2e 0f 1f 84 00 00 00
[42766.896527] RSP: 0018:ffff88803eb03af0 EFLAGS: 00000202 ORIG_RAX: ffffffffffffff13
[42766.896611] RAX: 0000000000007f9d RBX: ffff88802d42bb40 RCX: 0000000060cec985
[42766.896690] RDX: 0000000000000001 RSI: 0000000000007f9d RDI: ffff88802d42bb44
[42766.896769] RBP: ffff88803eb03af0 R08: 0000000000000000 R09: 0000000000000001
[42766.896847] R10: 0000000000000001 R11: 0000000060cec985 R12: ffff88802d42bb44
[42766.896926] R13: 0000000000000001 R14: ffff88803eb03b48 R15: ffff88802cac2c40
[42766.897006]  ? apic_timer_interrupt+0xa/0x20
[42766.897057]  _raw_spin_lock+0x1a/0x20
[42766.897107]  fast_classifier_recv+0x1798/0x1def [fast_classifier]
[42766.897177]  ? br_pass_frame_up+0x10f/0x120
[42766.897226]  ? nf_nat_inet_fn+0x181/0x280 [nf_nat]
[42766.897280]  fast_classifier_recv+0x1de1/0x1def [fast_classifier]
[42766.897351]  nf_hook_slow+0x36/0xa0
[42766.897392]  ip_output+0x7a/0xa0
[42766.897430]  ? __ip_flush_pending_frames.isra.50+0x80/0x80
[42766.897492]  ip_forward_finish+0x42/0x50
[42766.897537]  ip_forward+0x36d/0x3e0
[42766.897577]  ? ip4_key_hashfn+0xb0/0xb0
[42766.897622]  ip_rcv_finish+0x3a/0x40
[42766.897663]  ip_rcv+0x83/0x90
[42766.897698]  ? ip_rcv_finish_core.isra.13+0x250/0x250
[42766.897756]  __netif_receive_skb_one_core+0x43/0x50
[42766.897812]  __netif_receive_skb+0x54/0x60
[42766.897859]  process_backlog+0xad/0x150
[42766.897903]  net_rx_action+0xf3/0x270
[42766.897946]  __do_softirq+0xba/0x1f4
[42766.897989]  irq_exit+0x62/0x90
[42766.898026]  do_IRQ+0xaf/0xd0
[42766.898060]  common_interrupt+0xf/0xf
[42766.898102]  </IRQ>
[42766.898129] RIP: 0010:native_safe_halt+0x17/0x20
[42766.898181] Code: 07 0f 00 2d 3b 7c 58 00 f4 5d c3 0f 1f 84 00 00 00 00 00 8b 05 8a 63 b3 00 55 48 89 e5 85 c0 7e 07 0f 00 2d 1b 7c 58 00 fb f4 <5d> c3 90 90 90 90 90 90 90 55 48 89 e5 e8 57 46 a5 ff 65 8b 05 60
[42766.898384] RSP: 0018:ffffc9000007fea0 EFLAGS: 00000246 ORIG_RAX: ffffffffffffffd8
[42766.898468] RAX: 0000000000000000 RBX: 0000000000000001 RCX: ffff88803eb1a440
[42766.898547] RDX: ffffffff82032210 RSI: ffff88803eb1a440 RDI: 0000000000000000
[42766.898625] RBP: ffffc9000007fea0 R08: 0000000000000001 R09: 0000000000000001
[42766.898704] R10: 0000000000000001 R11: 0000000000000000 R12: 0000000000000000
[42766.898783] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[42766.898863]  default_idle+0x10/0x20
[42766.898904]  arch_cpu_idle+0x10/0x20
[42766.898946]  default_idle_call+0x27/0x30
[42766.898991]  do_idle+0xb3/0x180
[42766.899028]  cpu_startup_entry+0x5f/0x70
[42766.899073]  start_secondary+0x196/0x1a0
[42766.899118]  secondary_startup_64+0xa4/0xb0

Dimitri2020007, 2020-11-01:
你这issue咋和我的一样啊，都是这个错误无限刷屏。不过我是hyper-v下运行的虚拟机。固件已经装了docker和sfe+flowoffload

wx611975, 2020-11-02:
我也是这个错误

wx611975, 2020-11-02:
我是4415u

Dimitri2020007, 2020-11-02:
这个错误是不是老刷屏？一刷就停不下来的那种？我调过编译选项都没用啊。

Dimitri2020007, 2020-11-04:
在menuconfig>utilities下的>coreutils（the gnu core utilities）全部添加上。

看看挂起现象是不是恢复了。

添加这个可以少刷屏或者不刷屏。esxi6.7测试的结果，hyper-v也可以控制住刷屏。

Dimitri2020007, 2020-11-06:
删了acpica-unix试一下

coolsnowwolf, 2020-11-08:
sfe+flowoffloa 冲突了

Dimitri2020007, 2020-11-10:
已经确认了是sfe和flowoffload有冲突。建议sfe和flowoffload编译时候二选一，（推荐只装sfe）。已经解决了刷屏的问题。亲测有效。

closed 2021-06-20 by coolsnowwolf