coolsnowwolf/lede #5736: 编译后大于路由器可上传固件

z719893361, 2020-11-01:
我是一个小白，最新刚接触openwrt，想为手里的K2编译一个最新的固件，K2上传最大为7m源码编译后却有16M，请问该怎么设置才可以保持在7M之内

Cathgao, 2020-11-01:
减少开启的模块

Dimitri2020007, 2020-11-01:
减少固件kmod，（像menuconfig>kernel modules中的驱动，保留相应机型的有线\无线网卡驱动config中的firmware，network中的软件。utilties下的软件。精简luci-app！软件太长时间无新版本（半年以上未更新的luci-app），或功能重复的，（像smb/cifs/wifi/ssr服务端/qos/turbo acc/加速/dns部分的，smb/smb4/cifsd/cifs mount，dynapoint，guest-wifi，ss-libev，ssr-python（太长时间未更新版本。类似的ssr-mudb也是python，已支持最新的chacha加密，该源码未收录），qos与sqm，flowoffload与sfe，dnscrypt-proxy/dns-https-proxy/unbound（强烈推荐优先选unbound与dnsmasq一起用））碰到以上软件的二选一就好，还有一些是很少用到的，ahcp，hnet，bmx路由协议，bird路由协议，olsrd软件，cshark（作用不知道）以上是能精简的luci-app。

如果有强大的动手能力，可以找个容量更大的flash替换掉容量小的。则不需要再看精简软件列表了，可以想装多少就装多少。

z719893361, 2020-11-02:
OK，的确是插件占用太多。谢谢

closed 2020-11-02 by z719893361